ikarus23/MifareClassicTool #51: About galaxy s5 support for mifare (900p model, sprint)

deviker, 2015-01-27:
Galaxy s5 900P says it is not compatble with mifare classic. If it won't be just add to the documentation.

It did show some strings with generic info and card uid.

ikarus23, 2015-01-30:
Thanks for the notice. I will add it to the list.

Are you using Android 5.0? Because there are several people experiencing weird behavior by using MCT (or NFC in general) on an S5 with Android Lollipop.
* http://www.androidpit.com/galaxy-s5-android-update
* http://androidheadlines.com/2014/11/aosp-based-android-5-0-lollipop-builds-now-available-for-samsungs-galaxy-s4-and-s5.html

I'm not very familiar with the Galaxy S5 series. In what hardware does the 900P differ from a "regular" S5?

Can you install an app like [NFC Taginfo](https://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo&hl=en) and check if you can read any data from a Mifare Classic tag?

deviker, 2015-01-31:
Hi, the 900P is the cdma version of the S5 (it also has 3g, gsm, gprs
capabilities but it uses different hardware, different enough to have a
different cyanogen branch) .

I am using stock 4.4.2, not lollipop or cm12.

I'll tell you what I can read as soon as I get another card (monday
probably)

deviker, 2015-02-04:
Two screenshots with 2 mifare cards

ikarus23, 2015-02-14:
Done.
See: 31a218899a76111b92542409393c9aac7b327cc8

closed 2015-02-14 by ikarus23