ikarus23/MifareClassicTool #52: HTC M8 LOLLIPOP

torberglund, 2015-02-08:
MCT has always been working flawless on my HTC M8 pre-lollipop. This weekend I updated the device to 5.0,  since then MCT crashes immediately when reading or writing to a tag.

alierta, 2015-02-08:
I have the same problem. I will have to downgrade the phone

ikarus23, 2015-02-09:
Custom or official build?

alierta, 2015-02-09:
Official Lolipop (vodafone uk)

ikarus23, 2015-02-09:
Thanks for the info and - of course - thanks for reporting.
I just noticed that you tow are not the only ones. I got 8 crash reports via google play... The issues is within:
```
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'short android.os.Bundle.getShort(java.lang.String)' on a null object reference
at android.nfc.tech.NfcA.<init>(NfcA.java:92)
at android.nfc.tech.NfcA.get(NfcA.java:78)
at android.nfc.tech.MifareClassic.<init>(MifareClassic.java:166)
at android.nfc.tech.MifareClassic.get(MifareClassic.java:156)
at de.syss.MifareClassicTool.MCReader.<init>(MCReader.java:67)
at de.syss.MifareClassicTool.MCReader.get(MCReader.java:81)
at de.syss.MifareClassicTool.Common.checkForTagAndCreateReader(Common.java:622)
at de.syss.MifareClassicTool.Activities.KeyMapCreator.onCreateKeyMap(KeyMapCreator.java:372)
```
But I'm not really sure what is causing it. It occurs deep down in some Android method (as you can see from the `android.` package name. I will try to reproduce it order to find out more. Unfortunately I don't own a HTC M8, so I will try my best without one.

ikarus23, 2015-02-09:
Also: can you check if other apps work well?
e.g. [NFC TagInfo](https://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo) (you can set key A and B via the settings) or [NXP TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite)

alierta, 2015-02-09:
Yes I have tested it, The other apps doesn't work. Nfc war, NFC taginfo (NXP)..., and all I tested. The only app that works is NFC Tools (by wakdev). All apps says the same error.

If you need help for test changes contact with me, If i can help in something. (sorry for my english) and thanks

ikarus23, 2015-02-09:
Thanks for the testing!
So there is an issue within the Lollipop build that affects nearly all NFC applications. Now we have to find out what wakdev is doing better than all other NFC apps ;)

ikarus23, 2015-02-09:
Hmm... so this issue is connected with https://github.com/ikarus23/MifareClassicTool/issues/53

alierta, 2015-02-10:
Today I tested the MCT tool in LG g3 with Lollipop and works perfectly. Best regards

tinchogon34, 2015-02-12:
Same problem here with M8 and official lollipop :(

tinchogon34, 2015-02-12:
I tried to debug this error but it appears to be an Android nfc tech library bug.

ikarus23, 2015-02-12:
@tinchogon34 In Android's nfc tech library, you say... this sounds bad. But if some other apps work there must be a way...

I don't know if there is any difference, but I compiled a testing version based on the latest source with the latest Android Studio: http://tests.icaria.de/MifareClassicTool-1.8.3-testing.apk
(All releases prior this testing version were compiled by using the old adt eclipse bundle.)

Jawler, 2015-02-13:
I also have the same issue with Stock Rom. Wakdevs app only works to read the UID. If you want to read the data it says "Not supperted yet".
I tested with your testing version. Detecting the tag works but when i want to read data it crashes.

Jawler, 2015-02-13:
Also i tested 3 or 4 other apps. All the same issue. If they allow to read UID only it works fine. As soon as i want to read data the app crashes. So i guess it is a general Lollipop/HTC lollipop issue.

torberglund, 2015-02-20:
Any news on the matter?

ikarus23, 2015-02-21:
Not yet, sorry.

arrabal, 2015-02-26:
I just want to add that this exact issue is happening to HTC One M7 on Lollipop. It worked perfectly in 4.x and crashes when reading/writing after updating to 5.0.2. 
Also, tested some other apps as well and they all have the same problem.

Should I open a different issue for HTC One M7 or is it the same bug?

ikarus23, 2015-02-26:
Thanks for the Information. No need to open another issue because it is the same symptom (and most likely the same cause). I updated the title to be more clear.

DanielHosseini, 2015-03-12:
I have the same problem as arrabal, seems like lollipop is the evil in this :(

alfs, 2015-03-18:
Sony Z3 compact, Lollipop 5.0.2 (build 23.1.A.0.690) is working fine with MCT and NFC TagInfo.
So the bug may not be related to Android's nfc tech library in general, but to specific to certain phone models. I see another user was successful also with LG G3. I know it's not much comfort for M7/M8 owners, but may still be useful information for the developer.

mdanny82, 2015-03-30:
I have a HTC One M7. Same Issue after upgrading to Lollipop 5.0.2 (Stock Rom Vodafone Germany). If u need someone to make tests with apk-versions with stacktraces or need other Information, plz contact me.

ikarus23, 2015-04-02:
Hey guys,
I managed to contact HTC about this issue. To investigate this further they need some information. Unfortunately I don't own a HTC One (m7/m8). Can one of you help me out? They need:
* Serial Number:
* IMEI Number:
* Phone Model:
* Build Number:
* Android Version:
* Baseband Version:
* Sense Version:
* Current network provider:

If you don't want to post this information here feel free to write me a mail. You can get the address from the google play page of MCT.

Thanks for your help!

arrabal, 2015-04-05:
Email sent. I hope it helps. Thank you.

mdanny82, 2015-04-06:
Also sent an email.

    Serial Number: in Mail
    IMEI Number: in Mail
    Phone Model: HTC One
    Build Number: 7.18.161.2
    Android Version: 5.0.2
    Baseband Version: 4T.35.3218.16_10.33Q.1718.01L
    Sense Version: 6.0
    Current network provider: Vodafone Germany

TogetherRolle, 2015-04-07:
First of all thank you for making this software. Unable Chinese firewall in the city to pay onGoogle player feel very sorry. So I can only download pirated. I am a middle school student China, M8 I have encountered this problem, hope to be able to repair it as soon as possible. English is not good please do forgive.

TogetherRolle, 2015-04-07:
By the way, where you use M8 or HTC to do more people, in Chinese is not very popular.Thanks.

ikarus23, 2015-04-07:
@TogetherRolle you can download the "official" (created and signed by me) APK files [here](http://publications.icaria.de/mct/releases/). I hope this site is not blocked by the Chinese firewall.

ikarus23, 2015-04-09:
FYI (answer by HTC):
>Thank you for your reply and the requested information.

>I have forwarded this information to our escalation team and I will get back to 
>you after I get a reply from them.

GSvo, 2015-04-10:
Thx a lot ikarus23!

arrabal, 2015-04-17:
I am thankful for your efforts ikarus.  The day I upgraded to lollipop I contacted HTC as well and they told me they couldn't be sure if the error is caused by the firmware or the apps, even when I told them I had tried around 6 different apps from the store.  I'm talking about one month  ago.  They said an official HTC app must be used to determine whether the NFC  feature works or not but,  ironically, there is not such official app. At the end they recommended to wait until the developers troubleshooted this but I think that at this point this very clear it is their firmware's fault. 

I think you all would agree this is frustrating.  One waits for an OS upgrade for months and when it comes things stop working. To make it even worse,  the vendor announces there won't be more  updates for M7 and you end up regretting to upgrade your phone.  I  even asked HTC fora way to revert  to kit at,  but they said it is not possible. 

So, while they fix this,  that will take months if they finally do it,  do any of you know how to revert M7 to 4.4? Lollipop is rubbish anyway,  for its battery consumption and it's there is no way  to stop notifications vibration. I know this is a bit off topic,  but I am desesperated here and could not find help in forums. 

Thank you guys

ikarus23, 2015-04-17:
Hi arrabal,
thanks for sharing the information. Sadly, you're probably right. If we are lucky, this could take month... But maybe they take this more serious if more and more people complain about it!?

Regarding your question: Sorry, I'm not familiar with with downgrading HTC devices. :-/

arrabal, 2015-04-29:
Hey guys, I'm messaging with HTC twitter account and here's  the conversation me so far:

[image]

I can't try cache clearing right now because my phone is being repaired, so if anyone wants to try though I'm 99,99999999% sure that won't help. It seems there is lack of internal communication about this issue within HTC. However, companies often tend to be more supportive if you interact with their social network accounts than by email or phone. 

I encourage all of you to report this problem via twitter/faceboot/whatever, so they realize this is happening and they don't ignore us.

Thank you.

StHuginn, 2015-05-04:
@arrabal Hi, just did cache clearing on M7 lollipop - no good as expected.

ikarus23, 2015-05-09:
As I can see from the submissions of crash reports via Google Play, the **HTC One E8** (htc_mecul_emea) is affected by this issue as well....

dschuermann, 2015-05-11:
At OpenKeychain, we also have problems with HTC One and NFC, see our bug at https://github.com/open-keychain/open-keychain/issues/990

bildin, 2015-05-20:
Hi!
It seems, the reason of this bug in TechExtras of NfcA is null. However, TechList contains MifareClassic.

In one of my application I tried to patch tag adding own Extras if needed.

Here is the code:

	final public Tag patchTag(Tag oTag)
	{
		Parcel oParcel, nParcel;

		if (oTag == null) 
			return null;

		oParcel = Parcel.obtain();
		oTag.writeToParcel(oParcel, 0);
		oParcel.setDataPosition(0);

		int id_len = oParcel.readInt();
		byte[] id = null;
		if (id_len >= 0)
		{
			id = new byte[id_len];
			oParcel.readByteArray(id);
		}
		int[] oTechList = new int[oParcel.readInt()];
		oParcel.readIntArray(oTechList);
		Bundle[] oTechExtras = oParcel.createTypedArray(Bundle.CREATOR);
		int serviceHandle = oParcel.readInt();
		int isMock = oParcel.readInt();
		IBinder tagService;
		if (isMock == 0)
		{
			tagService = oParcel.readStrongBinder();
		}
		else
		{
			tagService = null;
		}
		oParcel.recycle();
		
		boolean isClassic = false;
		boolean isUltralight = true;
		
		for(int tech : oTechList)
		{
			switch(tech)
			{
				case 8:	//MifareClassic
					isClassic = true;
				case 2:	//NfcB
				case 3:	//IsoDep
				case 4:	//NfcF
				case 5:	//NfcV
				case 6:	//NDEF
				case 10:	//NfcBarcode
					isUltralight = false;
				default:
			}
		}
		
		Bundle oExtra, nExtra;
		short oSak, nSak;
		byte[] oAtqa, nAtqa;
		int[] nTechList = new int[oTechList.length];
		Bundle[] nTechExtras = new Bundle[nTechList.length];
		
		for(int idx = 0; idx < oTechList.length; idx++)
		{
			if(oTechList[idx] == 1)	//NfcA
			{
				oExtra = oTechExtras[idx];
				if(oExtra != null)
				{
					oSak = oExtra.getShort("sak");	//SAK
					oAtqa = oExtra.getByteArray("atqa");	//ATQA
					nExtra = oExtra;
				}
				else
				{
					if(isClassic)
					{
						nSak = 0x08;
						if(id_len == 4)
							nAtqa = new byte[]{(byte)0x04,(byte)0x00};
						else if(id_len == 7)
							nAtqa = new byte[]{(byte)0x44,(byte)0x00};
						else
							nAtqa = new byte[]{(byte)0x00,(byte)0x00};
					}
					else if(isUltralight)
					{
						nSak = 0x00;
						nAtqa = new byte[]{(byte)0x44,(byte)0x00};
					}
					else
					{
						nSak = 0x80;
						nAtqa = new byte[]{(byte)0x00,(byte)0x00};
					}
					
					nExtra = new Bundle();
					nExtra.putShort("sak",nSak);
					nExtra.putByteArray("atqa",nAtqa);
				}
				nTechExtras[idx] = nExtra;
			}
			else
			{
				nTechExtras[idx] = oTechExtras[idx];
			}
			nTechList[idx] = oTechList[idx];
		}

		nParcel = Parcel.obtain();
		nParcel.writeInt(id.length);
		nParcel.writeByteArray(id);
		nParcel.writeInt(nTechList.length);
		nParcel.writeIntArray(nTechList);
		nParcel.writeTypedArray(nTechExtras,0);
		nParcel.writeInt(serviceHandle);
		nParcel.writeInt(isMock);
		if(isMock==0)
		{
			nParcel.writeStrongBinder(tagService);
		}
		nParcel.setDataPosition(0);
		
		Tag nTag = Tag.CREATOR.createFromParcel(nParcel);

		nParcel.recycle();

		return nTag;
	}

usage example:

	Tag tag = intent.getParcelableExtra(NfcAdapter.EXTRA_TAG);
	if(tag!=null)
	{
		try
		{
			mNfcA = NfcA.get(tag);
		}
		catch(NullPointerException e)
		{
			try
			{
				tag = patchTag(tag);
				mNfcA = NfcA.get(tag);
			}
			catch(NullPointerException  e2)
			{
			}
		}
	}
	
But I have no device to test.

Could someone try it? Are there any other solutions?

iceiony, 2015-05-20:
@bildin patch code didn't work , but I was able to adapt it to a working solution.
The null pointer is thrown on MifareClassic.get(tag) , instead of NfcA.getTag() .
The bundle information does exist on the tag object, but it is in the wrong index in TechExtras[] . I had to find the correct index and simply assign that object to the desired tag index .

arrabal, 2015-05-20:
I can test the patch in my device tonight. Where can I download it from? If it's not published I think I can addapt the code myself following @iceiony indications

bildin, 2015-05-20:
@iceiony that's good news.

According to android sources Mifare Classic is always based on NfcA:

public static MifareClassic get(Tag tag) {
if (!tag.hasTech(TagTechnology.MIFARE_CLASSIC)) return null;
	try {
		return new MifareClassic(tag);
	} catch (RemoteException e) {
		return null;
	}
}

public MifareClassic(Tag tag) throws RemoteException {
	super(tag, TagTechnology.MIFARE_CLASSIC);
	NfcA a = NfcA.get(tag);  // MIFARE Classic is always based on NFC a

and NfcA in own order gets TechExtras from tag by TagTechnology.NFC_A (1):

public static NfcA get(Tag tag) {
	if (!tag.hasTech(TagTechnology.NFC_A)) return null;
		try {
			return new NfcA(tag);
		} catch (RemoteException e) {
			return null;
		}
	}

public NfcA(Tag tag) throws RemoteException {
	super(tag, TagTechnology.NFC_A);
	Bundle extras = tag.getTechExtras(TagTechnology.NFC_A);
	mSak = extras.getShort(EXTRA_SAK);
	mAtqa = extras.getByteArray(EXTRA_ATQA);
}
	
getTechExtras function in Tag class is next:
	
public Bundle getTechExtras(int tech) {
	int pos = -1;
	for (int idx = 0; idx < mTechList.length; idx++) {
		if (mTechList[idx] == tech) {
			pos = idx;
			break;
		}
	}
	if (pos < 0) {
		return null;
	}
	return mTechExtras[pos];
}
	
So in the right order Extras should be at the index of NfcA tech (int 1) in TechList[].

Where did you find these Extras finally? And what index did you assign them?

iceiony, 2015-05-21:
@bildin  it was as you say, at index 1 corresponding to NfcA. I had copied the techExtra to index 0 .

After investigating a bit further , the debugger shows that HTC One sources don't match the android 21 code .

Most importantly , TagTechnology.NFC_A returns 0 instead of 1

Here's the change I did to your code :

```java
        int[] nTechList = new int[oTechList.length];
        Bundle[] nTechExtras = new Bundle[nTechList.length];
        int bundleIndex;
        for(bundleIndex=0;bundleIndex<oTechExtras.length;bundleIndex++){
            if(oTechExtras[bundleIndex]!=null){
                break;
            }
        }

        for(int idx = 0; idx < oTechList.length; idx++)
        {
            if(oTechExtras[idx] == null)
            {
                nTechExtras[idx] = oTechExtras[bundleIndex];
            }
            else
            {
                nTechExtras[idx] = oTechExtras[idx];
            }

            nTechList[idx] = oTechList[idx];
        }
```

bildin, 2015-05-21:
@iceiony thank you.

And if tech at at index 0 is MifareClassic, I think it is possible to simplify and make code more common:

	public Tag patchTag(Tag oTag)
	{
		if (oTag == null) 
			return null;
			
		String[] sTechList = oTag.getTechList();
			
		Parcel oParcel, nParcel;
		
		oParcel = Parcel.obtain();
		oTag.writeToParcel(oParcel, 0);
		oParcel.setDataPosition(0);

		int len = oParcel.readInt();
		byte[] id = null;
		if (len >= 0)
		{
			id = new byte[len];
			oParcel.readByteArray(id);
		}
		int[] oTechList = new int[oParcel.readInt()];
		oParcel.readIntArray(oTechList);
		Bundle[] oTechExtras = oParcel.createTypedArray(Bundle.CREATOR);
		int serviceHandle = oParcel.readInt();
		int isMock = oParcel.readInt();
		IBinder tagService;
		if (isMock == 0)
		{
			tagService = oParcel.readStrongBinder();
		}
		else
		{
			tagService = null;
		}
		oParcel.recycle();
		
		int nfca_idx=-1;
		int mc_idx=-1;
		
		for(int idx = 0; idx < sTechList.length; idx++)
		{
			if(sTechList[idx] == NfcA.class.getName())
			{
				nfca_idx = idx;
			}
			else if(sTechList[idx] == MifareClassic.class.getName())
			{
				mc_idx = idx;
			}
		}
		
		if(nfca_idx>=0&&mc_idx>=0&&oTechExtras[mc_idx]==null)
		{
			oTechExtras[mc_idx] = oTechExtras[nfca_idx];
		}
		else
		{
			return oTag;
		}

		nParcel = Parcel.obtain();
		nParcel.writeInt(id.length);
		nParcel.writeByteArray(id);
		nParcel.writeInt(oTechList.length);
		nParcel.writeIntArray(oTechList);
		nParcel.writeTypedArray(oTechExtras,0);
		nParcel.writeInt(serviceHandle);
		nParcel.writeInt(isMock);
		if(isMock==0)
		{
			nParcel.writeStrongBinder(tagService);
		}
		nParcel.setDataPosition(0);
		
		Tag nTag = Tag.CREATOR.createFromParcel(nParcel);

		nParcel.recycle();

		return nTag;
	}

freerider7777, 2015-05-21:
@bildin 
Super, works fine!!! Thank you.

freerider7777, 2015-05-21:
@bildin 
BTW, what if we change 
TagTechnology.NFC_A value to 1 (using reflection) ?

bildin, 2015-05-22:
@freerider7777 I suppose based on users` reports from my app that TagTechnology.NFC_A is 1.
I was able to dump TechList in String and in Int (in brackets) and compare them. Also I checked TechExtras.
Some results a below:

___
Android 5.0.1; HTC One_M8 Build/LRX22C;

0:android.nfc.tech.MifareClassic(8)
 Extra: null
1:android.nfc.tech.NfcA(1)
 Extra: Bundle[{sak=8, atqa=4400}]
2:android.nfc.tech.NdefFormatable(7)
 Extra: null

___
Android 5.0.2; HTC One Build/LRX22G;

0:android.nfc.tech.MifareClassic(8)
 Extra: null
1:android.nfc.tech.NfcA(1)
 Extra: Bundle[{sak=8, atqa=4400}]
2:android.nfc.tech.NdefFormatable(7)
 Extra: null

___
Android 5.0.2; HTC One dual sim Build/LRX22G;
 
0:android.nfc.tech.MifareClassic(8)
 Extra: null
1:android.nfc.tech.NfcA(1)
 Extra: Bundle[{sak=8, atqa=4400}]
2:android.nfc.tech.NdefFormatable(7)
 Extra: null

___
Android 5.0.2; HTC One_E8 dual sim Build/LRX22G;
 
0: android.nfc.tech.MifareClassic = 8
 Extra: null
1: android.nfc.tech.NfcA = 1
 Extra: Bundle[{sak=8, atqa=4400}]
2: android.nfc.tech.NdefFormatable = 7
 Extra: null

___
Android 5.0.1; HTC One_M8 Build/LRX22C;

0:android.nfc.tech.IsoDep(3)
 Extra: Bundle[{histbytes=byte[]}]
1:android.nfc.tech.MifareClassic(8)
 Extra: null
2:android.nfc.tech.NfcA(1)
 Extra: Bundle[{sak=56, atqa=0200}]
3:android.nfc.tech.NdefFormatable(7)
 Extra: null

As you can see, sometimes at index 0 is IsoDep technology, but no crashes due to copying NfcA Extra to MifareClassic`s index.

freerider7777, 2015-05-22:
@bildin 
Ok! Thanks for this info. So TagTechnology.NFC_A value is constant.

As far as I understand in first 4 cases MifareClassic works without need to patch?

As we see in the code of NFCA (it's called inside MifareClassic), causing "java.lang.NullPointerException"
 public NfcA(Tag tag) throws RemoteException {
        super(tag, TagTechnology.NFC_A);
        Bundle extras = tag.getTechExtras(TagTechnology.NFC_A);
        mSak = extras.getShort(EXTRA_SAK);
        mAtqa = extras.getByteArray(EXTRA_ATQA);
    }

it takes Extras from TagTechnology.NFC_A index, not from mifare technology index :)

In 5th case they just match (we're lucky)
mc_idx = 1 (in the code)
and
TagTechnology.NFC_A = 1

But if mifare moves to other index it'll fail...

bildin, 2015-05-22:
@freerider7777 all of this cases are from patch execution, which is beeing called only if crash has occured. In all of this cases no crash after executed patch.

freerider7777, 2015-05-22:
Ok! Cool then :)

bildin, 2015-05-22:
@freerider7777 btw, my app connects to Tag with NfcA first, even to MifareClassic tag. And it also crashed before patch.

freerider7777, 2015-05-22:
@bildin Very strange... the constants TagTechnology.NFC_A, TagTechnology.* are the same as in original code. Seems some indexing bug in HTC build...

iceiony, 2015-05-22:
@freerider7777 Like I mentioned, the sources don't seem to match. Regarless of what you see in the android sdk , HTC one seems to be running slightly different code.

freerider7777, 2015-05-22:
@iceiony Yes, I've seen your post... I was thinking about embedding original google code in apk to try to fix this problem, but didn't have time to try this. Now with this patch there is no need! :)

iceiony, 2015-05-22:
@freerider7777 Oooh . Sorry for missunderstanding :D

ikarus23, 2015-05-22:
Wow! So you're telling me if I implement [the patch by bildin](https://github.com/ikarus23/MifareClassicTool/issues/52#issuecomment-104277445) the issue will be gone on HTC One m7/m8 devices? For real?

I will try to publish a testing version with this patch applied as soon as possible (unfortunately not before May 25th).

freerider7777, 2015-05-22:
@ikarus23 
yep :) tested on several HTC devices with Android 5.*

ikarus23, 2015-05-22:
Ok, I was too excited to do anything else... ;)
Please download an try [this testing version of MCT](http://tests.icaria.de/MifareClassicTool-2.0.2-testing.apk).

Jawler, 2015-05-22:
It is working. Guess you are the only one having a working Nfc reader cor One's on 5.* now.

freerider7777, 2015-05-22:
@bildin What's your app? :) If not a secret... You made a lot of people happy with this patch :))

ikarus23, 2015-05-22:
You guys are awesome!
Thank you. All of you. And especially bildin for analysing this further and providing a patch!

P.S.:Don't forget to use `equals()` to compare stings in Java. ;)

freerider7777, 2015-05-22:
@ikarus23 oh yes... you mean to replace these "ifs":
 if(sTechList[idx] == NfcA.class.getName())
which can fail to compare...

arrabal, 2015-05-22:
the testing version works on HTC one m7. Amazing! Thank you guys, this os awesome

[15 more replies]

closed 2015-05-24 by ikarus23