ikarus23/MifareClassicTool #53: s5 on lolipop showing dashes

rich330, 2015-02-09:
When ever i try and read a mifare card key a is visible but everything else has dashes

ikarus23, 2015-02-09:
This is a well known issue with custom Lollipop builds.
* http://www.androidpit.com/galaxy-s5-android-update
* http://www.androidheadlines.com/2014/11/aosp-based-android-5-0-lollipop-builds-now-available-for-samsungs-galaxy-s4-and-s5.html

Are you using an official build or a custom one?

And you are not alone. I received a lot of messages from peoples with Lollipop on a S5. Until now I don't know what is causing this issue. I don't own a S5. Maybe one of my friends has one I can borrow for debugging purposes.

ikarus23, 2015-02-09:
Also: can you check if other apps work well?
e.g. [NFC TagInfo](https://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo) (you can set key A and B via the settings) or [NXP TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite)

kli44, 2015-02-09:
I have the same issue with my S5 in lollipop, with NFC TAGINFO it's the same issue but RFID NFC Tool work fine.

ikarus23, 2015-02-09:
Hmm... so this issue is connected with https://github.com/ikarus23/MifareClassicTool/issues/52

rich330, 2015-02-10:
I am using an official build. NFC taginfo says read failed when checking the data on a brand new 1k mifare card. Using the same card NXP taginfo works ok.

Date: Mon, 9 Feb 2015 12:59:50 -0800

AnArchoz, 2015-02-13:
I get the same problem, official lollipop S5. But also when I try to write a previously working clone dump I get an error saying "Invalid Access Conditions or dead sector :(" in all 16 sectors. This never happened before, and tag is fully functioning. I suppose it's related to the reading issue.

AnArchoz, 2015-02-15:
Won't be able to report further on the lollipop build as I in my panic downgraded back to kitkat.

andake, 2015-03-28:
I tried this out on my S5 and its the MFC readBlock() method returning 17 bytes that causes this.
However, the additional byte is only 0x0. So by removing the 17th byte from the readBlock() return this issue was not seen anymore. If this is the right way to go I can't tell.

ikarus23, 2015-03-29:
Thank you for debugging this, @andake. I will create a testing version with this modification.
Oh, and stupid S5 rom... why 17 byte... the [spec](https://developer.android.com/reference/android/nfc/tech/MifareClassic.html#readBlock%28int%29) clearly stats always 16 bytes. I noticed that sometimes this method returns less than 16 bytes. This is why a added [this](https://github.com/ikarus23/MifareClassicTool/blob/master/Mifare%20Classic%20Tool/app/src/main/java/de/syss/MifareClassicTool/MCReader.java#L192).

viliny, 2015-03-30:
Hey, any closer to solving this?
Everything was working great until i upgraded to lollipop on my S5 Active - that 17 byte thingie looks like the issue here too - i tried "NFC Tag Cloner" (sheep logos everywhere if you try to find the app) and i get errors of tag data exceeding target capacity -> 1024 / 1072 

So looks like some data is made up somewhere along the way

ikarus23, 2015-03-30:
Hi,
I created a [testing version of MCT](http://tests.icaria.de/MifareClassicTool-2.0.0-testing.apk) with the changes @andake suggested. Please try it.

viliny, 2015-03-30:
Ok!

Reading now works, did you apply the same logic to writing somehow?

I just tried to write to a card that should allow sector 0 to be written to directly, but it just gives me a vague error message about some errors having occured - uid seems unchanged.

ikarus23, 2015-03-30:
No, I did not apply the same logic to writing. It is not possible. I'm pretty sure there is something wrong with your card. Before you try fancy things like writing to block 0 you just can try a simple write to an other block (as long you got the sufficient rights).

rich330, 2015-03-30:
Reading and writing worked for me. Thanks

ikarus23, 2015-03-30:
Thanks for the feedback. I will release this fix tomorrow with version 2.0.1.

AnArchoz, 2015-03-30:
Awesome, I'm gonna try it tomorrow!

viliny, 2015-03-31:
Thank you ikarus :)

mdanny82, 2015-03-31:
@ikarus23 testing Version still not working on my HTC One M7

ikarus23, 2015-03-31:
The fix (workaround, 6412ec60d7e9f47facc793a06c606d5ffbb8cf93) is now in the master branch and was released with version 2.0.1.

closed 2015-03-31 by ikarus23