ikarus23/MifareClassicTool #61: Enable support for Xperia Z2 (D6543)

DiegoQueiroz, 2015-05-04:
Currently, it is not possible to install MCT from the Play Store due to compatibility issues, but Xperia Z2 is able to read Mifare Classic tags and, therefore, is fully compatible with this application.

I installed the APK on my Xperia Z2 (D6543) and it is working perfectly. So I think it should be enabled on the Play Store.

DiegoQueiroz, 2015-05-04:
It is important to notice that the Donate Version is already listed as compatible with Xperia Z2.

ikarus23, 2015-05-04:
Are you sure? Have you verified reading and writing works? A while ago some users reported that MCT does not work on Z2. [This site](http://www.shopnfc.it/en/content/7-nfc-device-compatibility) says so too.

As I can see in the Google Play Developer interface, there are a bunch of Z2 out there. What exact version do you got there?

A list of Z2 devices:
* SonyXperia Z2 Tablet– SGP512
* SonyXperia Z2 Tablet– SGP511
* SonyXperia Z2 Tablet– SGP551
* SonyXperia Z2– D6503
* SonyXperia Z2 Tablet– SO-05F
* SonyXperia Z2– D6502
* SonyXperia Z2a– D6563
* SonyXperia Z2 Tablet– SGP521
* SonyXperia Z2– SO-03F
* SonyXperia Z2– D6543
* SonyXperia Z2 Tablet– SGP561
* SonyXperia Z2 Tablet– SOT21

ikarus23, 2015-05-04:
Ups, sorry, I overlooked the **D6543** you mentions twice...
I will activate this model in the app store.

ikarus23, 2015-05-04:
Do you know if any other model of the list support Mifare Classic?

closed 2015-05-04 by ikarus23

DiegoQueiroz, 2015-05-04:
Interesting to know. So maybe some Xperia X2 models does not have NXP chips, while some have it.

Just terrible. ;/