ikarus23/MifareClassicTool #64: MCT and Sony Z3

ikarus23, 2015-06-06:
moscowneversleeping said:
>Hi. New version don't works with some type of Classic Tags. SONY Z3. Android 5.0.2.
I see debug information today, and write here crashing logs.

ikarus23, 2015-06-06:
Hi. Another user told me that the Sony Z3 does not support Mifare Classic. Are you sure it worked before? Have you verified this by downgrading to e.g. 2.0.1?.

ikarus23, 2015-06-06:
Ok, there are other users reporting [issues with using Sony's Z3 in combination with Mifare Classic tags](https://stackoverflow.com/questions/30238152/tag-incorrectly-enumerated-as-mifare-classic-sak-32).

First, Samsung's Galaxy S5 returns arrays with a length different from 16 bytes to a [readBlock()](https://developer.android.com/reference/android/nfc/tech/MifareClassic.html#readBlock%28int%29) (e.g. 6412ec60d7e9f47facc793a06c606d5ffbb8cf93) call, then HTC's One m7/m8 devices [deliver broken Tag objects](https://github.com/ikarus23/MifareClassicTool/issues/52) to the apps and now Sony's Z3 has Mifare Classic issues too?! Hmpf. Why do all manufacturers tamper with the NFC stack these days...

bildin, 2015-06-06:
@ikarus23 finally, issue with HTC is not in broken Tag. But NfcA.get(Tag) tries to get Extras from index of MifareClassic when it is enumerated in Techlist.
Applying idea of that patch to this issue, if it is realy so as mentioned in issue on stackoverflow linked above, it is possible to check if MifareClassic is enumerated in Techlist, then change SAK in Extras if is needed. But it may cause some troubles with identifying memory size, I think.
Manufacturers are improoving stack, as they think. Russian proverb tells "best" is the enemy of "good". Other engineers say "Don't touch while it works" :)

ikarus23, 2015-06-06:
Regarding the issue: Yeah, I thought so too. But I don't like to implement a patch every time a vendor changes the Android code. But I guess I have no choice if I want to have happy users...
Anyway, lets try to find out first whether the stackoverflow issue applies to this or not.

Off-Topic: Oh, so it's not the Tag object that is broken, it is the `NfcA.get()` method! (or is it the "unsorted" Techlist list withing the Tag object?)

moscowneversleeping, 2015-06-06:
Now im debugging MCT 2.0.4 on Sony z3, Android 5.0.2
See my logs here https://db.tt/xRXfg596
I thing that most important part is

06-06 20:08:09.031    1643-2355/? D/NfcAdaptation﹕ NfcAdaptation::HalDeviceContextDataCallback: len=4
06-06 20:08:09.031    1643-2334/? I/BrcmNfcNfa﹕ NFC received rsp gid:1
06-06 20:08:09.031    1643-2334/? I/BrcmNfcNfa﹕ nfa_dm_disc_discovery_cback (): event:0x4003
06-06 20:08:09.031    1643-2334/? I/BrcmNfcNfa﹕ nfa_dm_disc_sm_execute (): state: W4_HOST_SELECT (3), event: SELECT_RSP(4) disc_flags: 0x9
06-06 20:08:09.031    1643-2334/? I/BrcmNfcNfa﹕ nfa_dm_disc_sm_execute (): new state: W4_HOST_SELECT (3), disc_flags: 0x9
06-06 20:08:09.033    7531-7531/? E/AndroidRuntime﹕ FATAL EXCEPTION: main
    Process: de.syss.MifareClassicTool, PID: 7531
    java.lang.IllegalStateException: Could not execute method of the activity

ikarus23, 2015-06-06:
I can see there a `BrcmNfcNfa` which refers to Broadcom, a NFC chip manufacturer. This is most likely the proof that the Z3 uses a NFC controller by Broadcom. Unfortunately Broadcom chips have no Mifare Classic support. Please read: https://github.com/ikarus23/MifareClassicTool/issues/1. As far as I can see there is nothing I can do about it, sorry.

moscowneversleeping, 2015-06-06:
But MCT can read and wrute some tags
[image]

ikarus23, 2015-06-06:
Ok, this is weird...
Can you use the Tag Info tool of MCT to get the SAK value?

moscowneversleeping, 2015-06-06:
oh shit, I just tried to read the difficult tag and once again she read!
but the next time the application again fell.
Each time MCT showed tags UID
But MCT can't show "Display Tag Info" - failed
TagInfo Failed TOO,

But NFC Tools 3.8.1 Free Version about "wakdev" work correctly every one.
And i can show you all informatoin. (see photo)
Now I tested MCT 2.0.4 on Sony Z1 compact Android 5.0.2 whith this tags, and all wokrs!
I can do all other test (may be build MCT from source and debug in real time..)

[image]

moscowneversleeping, 2015-06-06:
Soon I will write test results MCT on Sony Z3 with early 4.4.2 Android

bildin, 2015-06-06:
@moscowneversleeping did you tried to read this card on another device? what kind of card is it? try to get info with https://play.google.com/store/apps/details?id=com.nxp.taginfolite

there is some important information in your log:

...
06-06 20:08:07.536    1643-2334/? D/BrcmNfcJni﹕ NfcTag::discoverTechnologies (discovery): enter: rf disc. id=2; protocol=128, mNumTechList=2
06-06 20:08:07.536    1643-2334/? D/BrcmNfcJni﹕ NfcTag::discoverTechnologies (discovery): index=0; tech=3; handle=1; nfc type=4
06-06 20:08:07.536    1643-2334/? D/BrcmNfcJni﹕ NfcTag::discoverTechnologies (discovery): index=1; tech=1; handle=1; nfc type=4
06-06 20:08:07.536    1643-2334/? D/BrcmNfcJni﹕ NfcTag::discoverTechnologies (discovery): index=2; tech=8; handle=2; nfc type=128
06-06 20:08:07.536    1643-2334/? D/BrcmNfcJni﹕ NfcTag::discoverTechnologies (discovery): index=3; tech=1; handle=0; nfc type=0
06-06 20:08:07.536    1643-2334/? D/BrcmNfcJni﹕ NfcTag::discoverTechnologies (discovery); mNumDiscTechList=4
06-06 20:08:07.536    1643-2334/? D/BrcmNfcJni﹕ NfcTag::discoverTechnologies (discovery): exit
...
06-06 20:08:07.544    1643-2334/? I/BrcmNfcNfa﹕ RW_SetActivatedTagType protocol:4, technology:0, SAK:32
...

Mifare Classic is enumerated in TechList (tech=8) and uses protocol (nfc type = 128), but tag is activated wth protocol 4 coresponding to IsoDep (tech=3)

May be your card is Mifare Plus switched to another security level?

ikarus23, 2015-06-06:
The screenshot of NFC Tools is showing a SAK of `0x20` (also present in the logs; `0x20` = 32). According to [NXP's guidelines on identifying Mifare tags](http://www.nxp.com/documents/application_note/AN10833.pdf) (Page 11), this a Mifare Plus ore Mifare DESFire tag.

moscowneversleeping, 2015-06-07:
I'm using a MIFARE Classic emulated
Look I find my problem
http://stackoverflow.com/questions/30238152/tag-incorrectly-enumerated-as-mifare-classic-sak-32

Can you build test version? where MCT dont see on SAK and application will be use direct commands for reading.  (As you do for write zero block may be)

moscowneversleeping, 2015-06-07:
To day I installed Android 4.4.4 instead 5.0.2 on my Sony Z3 and run MCT
[image]

But yesterday I really was able to read the MIFARE Classic emulated Tag once (On Sony Z3). I know that this is possible, because Tag is reading on other phones (Sony Z1 and Nexus S, etc i'm testad it).

ikarus23, 2015-06-07:
*Regarding the stackoverflow issue*: I know this issue, I posted it right at the top of this issue ;)

*Regarding the testing version*: No, I can't build a version like this. On Android it is not possible to do real Mifare Classic commands. (Thats why only some of the UID changeable tags work).

*Regarding the screenshot*: MCT looks at the `TechList` of a tag. If there is no Mifare Classic it means that Android is not able to talk the Mifare Classic protocol with this tag. This could have two reasons. Either your device does not support Mifare Classic or your tag is no Mifare Classic tag. MCT tries to find out this reason by parsing several information. This includes the SAK. Because the SAK is `0x20` (which is according to [this guide](http://www.nxp.com/documents/application_note/AN10833.pdf) not a Mifare Classic tag) MCT is showing you the message "Not a Mifare Classic Tag".

moscowneversleeping, 2015-06-07:
Then I must modify the MifareClassic.java for correct display TechList information.
Can you recommend something else?
And you have any idea why once turned to consider the tag using MCT (read all sections with different keys)?

ikarus23, 2015-06-07:
As I said. The real fix (from an Android perspective) would be to change the SAK of the tag.
A dirty fix (which I will not implement) would be to patch the `TechList` of the tag to make Android think that this tag is able of talking Mifare Classic.

ikarus23, 2015-06-07:
Oh, that is weird. In the screenshot you've taken from NFC Tools, there is a Mifare Classic in the `TechList`. Did you do this screenshot with a different device? Or a different tag? Or was it just the newer Android version?

moscowneversleeping, 2015-06-07:
No. It's my Sony Z3, but Android 4.4.4 (then i install to day). I can Install 5.0.2 and look Tagingo NXP.

If nothing happens, I'll be grateful to you if you tell me a place in the source code to patch the TechList of the tag to make Android think that this tag is able of talking Mifare Classic.

moscowneversleeping, 2015-06-07:
More interesting sniffing information when MCT detectable Tag solve the information about it, and showing the message "Not a Mifare Classic Tag". https://cloud.githubusercontent.com/assets/2129699/8023350/ac7aa33a-0d11-11e5-8910-ead5b1b801a9.jpg

results of 
proxmark3> hf 14a snoop
https://db.tt/F9SLqeYS

bildin, 2015-06-07:
@moscowneversleeping it is strange thing in TechList NfcA occures twice. Let me guess that first (which is also returned by Tag object) contains SAK = 0x20, while second contains 0x18. It may caused by dual mode of this card. So I can try to compile test app for dumping and editing TechList tomorrow. Please contact me.

moscowneversleeping, 2015-06-07:
@bildin it would be perfect. As i can contact you?
my mail moscowvet@gmail.com

ikarus23, 2015-06-07:
*Regarding the screenshots from [this post](https://github.com/ikarus23/MifareClassicTool/issues/64#issuecomment-109733383)*: As you can see, the Mifare doctor app failed to read this tag. So does MCT, but it crashes from this failed read attempt. This is a thing I should definitively fix. NFC tool does not show any error because it has not tried to read the tag yet. All the information you can see in the screenshot are handed over to the app by Android.  The app doesn't need to read anything to display those information.

bildin, 2015-06-08:
@ikarus23 my assumption has been prooved...
Techlist contains:
0:IsoDep with histbytes in Extra
1:NfcA with SAK = 0x20
2:NfcA with SAK = 0x18 (at another card with same issue 0x08)
3:MifareClassic with the same Extra as at index 2 (earlier Android had been put null here, so therefor in previous issue HTC reads Extras from this index)
4:NdefFormatible with null in Extra.

I did OR both SAKs and put new Extra to the first occurance of NfcA. It works fine, but it makes me crazy to follow all of these variations.

moscowneversleeping, 2015-06-08:
@bildin Thanks for the help
@ikarus23 Can you make a changes to the main app code to read/write all available Mifare Classic tags in TehList?
This trend will continue in Android 5.* and it will be work with all new phones Sony (Z3 +, Z4 etc)

ikarus23, 2015-06-08:
Thanks again, @bildin, for bringing light into the into the dark corners of strange Android Mifare Classic issues! I'm not sure if I got you right on how to fix this. You added a new `Extra` to the `NfcA` at index 1? And removed the old one?) With a SAK of `0x20 | 0x18`?

Do you think there is a more general approach to fix these issues? Maybe manipulating the list to only contain one `MifareClassic` and one `NfcA` entry? If there are only those two, will the `MifareClassic.get(tag)` method work all the time?

@moscowneversleeping Lets try to find a general way to fix the TechList Extras that hopefully will work  for all devices with issues (e.g. the Sony Z3 and the HTC One). I would relay like to implement something like this and not just another fix for another device.

moscowneversleeping, 2015-06-08:
@ikarus23 Thanks. For my part, you can always count on quick tests of your application and any other help that I can provide.

bildin, 2015-06-09:
@ikarus23 I think more general Tag's cleanup will be something like this (in case of multi NfcA and to preserving functionality with other technologies)

	private Tag cleanupTag(Tag oTag) {
		if (oTag == null)
			return null;

		String[] sTechList = oTag.getTechList();

		Parcel oParcel = Parcel.obtain();
		oTag.writeToParcel(oParcel, 0);
		oParcel.setDataPosition(0);

		int len = oParcel.readInt();
		byte[] id = null;
		if (len >= 0) {
			id = new byte[len];
			oParcel.readByteArray(id);
		}
		int[] oTechList = new int[oParcel.readInt()];
		oParcel.readIntArray(oTechList);
		Bundle[] oTechExtras = oParcel.createTypedArray(Bundle.CREATOR);
		int serviceHandle = oParcel.readInt();
		int isMock = oParcel.readInt();
		IBinder tagService;
		if (isMock == 0) {
			tagService = oParcel.readStrongBinder();
		} else {
			tagService = null;
		}
		oParcel.recycle();

		int nfca_idx = -1;
		int mc_idx = -1;
		short oSak = 0;
		short nSak = 0;

		for (int idx = 0; idx < sTechList.length; idx++) {
			if (sTechList[idx].equals(NfcA.class.getName())) {
				if (nfca_idx == -1) {
					nfca_idx = idx;
					if (oTechExtras[idx] != null
							&& oTechExtras[idx].containsKey("sak")) {
						oSak = oTechExtras[idx].getShort("sak");
						nSak = oSak;
					}
				} else {
					if (oTechExtras[idx] != null
							&& oTechExtras[idx].containsKey("sak")) {
						nSak = (short) (nSak | oTechExtras[idx].getShort("sak"));
					}
				}
			} else if (sTechList[idx].equals(MifareClassic.class.getName())) {
				mc_idx = idx;
			}
		}

		boolean modified = false;

		if (oSak != nSak) {
			oTechExtras[nfca_idx].putShort("sak", nSak);
			modified = true;
		}

		if (nfca_idx != -1 && mc_idx != -1 && oTechExtras[mc_idx] == null) {
			oTechExtras[mc_idx] = oTechExtras[nfca_idx];
			modified = true;
		}

		if (!modified) {
			return oTag;
		}

		Parcel nParcel = Parcel.obtain();
		nParcel.writeInt(id.length);
		nParcel.writeByteArray(id);
		nParcel.writeInt(oTechList.length);
		nParcel.writeIntArray(oTechList);
		nParcel.writeTypedArray(oTechExtras, 0);
		nParcel.writeInt(serviceHandle);
		nParcel.writeInt(isMock);
		if (isMock == 0) {
			nParcel.writeStrongBinder(tagService);
		}
		nParcel.setDataPosition(0);

		Tag nTag = Tag.CREATOR.createFromParcel(nParcel);

		nParcel.recycle();

		return nTag;
	}

moscowneversleeping, 2015-06-26:
@ikarus23
Do you have any news about this issue ?

ikarus23, 2015-06-26:
Sorry moscowneversleeping, I'm really short on spare time right now. I will try to look into this deeper soon. However, I compiled a [testing version of MCT](http://tests.icaria.de/MifareClassicTool-2.0.4-testing.apk). This version implements the patch of bildin (just copy&paste). Please try it.

moscowneversleeping, 2015-06-27:
Yes, this patch work good! Thanks

ikarus23, 2015-06-30:
@moscowneversleeping: Great! Thanks for testing.
@bildin: If I got the patch right it will fix the Sony Z3 issue **and** the HTC One issue, right?

bildin, 2015-06-30:
@ikarus23, yes, You've got right. Notice, Sony Z3 issue doesn't call exception, so this modification of patch is manipulating with tech list forcefully for all devices to avoid occurrence of these issues on devices with need, but is harmless for devices with no need.

ikarus23, 2015-06-30:
@bildin Thanks for making this clear. And thanks again for another patch ;)

ikarus23, 2015-06-30:
@bildin Just another short question: Is there a reason you `or`ed the SAK values of the NfcA? Or did you just tested it with 52 as SAK and it worked? Do you think it will work for 24 or 32 as SAK too? Do you think it will work all the time if there only is one NfcA?

bildin, 2015-06-30:
@ikarus23 based on http://www.nxp.com/documents/application_note/AN10833.pdf pages 10-12:

3.2.1 Coding of SAK for MIFARE Implementation
In case of MIFARE Implementation, final SAK shall be set according to ISO14443-3 and
MIFARE SAKs. In case of multi-MIFARE implementation all supported SAKs can be
ORed to generate a SAK to be presented.

bit 4 responds to Mifare Classic (1K as default)
bit 1 - mini 0.3K
bit 5 - 4K
bit 6 - ISO1443-4 (in case of @moscowneversleeping)

so, as we tested with @moscowneversleeping 0x20 | 0x08 = 0x28 and 0x20 | 0x18 = 0x38 work well.

In case of only one NfcA oSak == nSak and patch doesn't touch SAK in extra.

I' m not sure what did you mean with SAK 52? It is mistype of 56, isn't it?

ikarus23, 2015-07-01:
@bildin Thanks for clearing that up! And of course I meant 56, not 52 :)

jlanza, 2015-07-02:
@bilding I have checked the code you provided in our application and the tag is detected and SAK is the correct one. But the problem is that when we try to read the memory from the MIFARE card we got an exception that the tag is lost. Any idea?

ikarus23, 2015-07-02:
I have created [another testing version of MCT](http://tests.icaria.de/MifareClassicTool-2.0.4-testing.apk) with a slightly modified version of bildin's patch (just code structure). Could somebody please test if this still works for a Sony Z3?

moscowneversleeping, 2015-07-02:
@ikarus23 
Sure! One minute;)

moscowneversleeping, 2015-07-02:
@ikarus23 All right!

moscowneversleeping, 2015-07-02:
No) I can Read, but i can not see Tag Info

ikarus23, 2015-07-02:
Thanks for testing! Were you able to display the Tag Info with the [first testing version](https://github.com/ikarus23/MifareClassicTool/issues/64#issuecomment-115902855)?

moscowneversleeping, 2015-07-02:
failed too (

ikarus23, 2015-07-02:
Thanks again.
Ok, as far as I remember the Tag Info tool does not patch the tag object. I think this could cause the issue. I will look into this and be back with another testing version as soon as possible (sorry, not today).

moscowneversleeping, 2015-07-02:
See you

ikarus23, 2015-07-04:
I've created [another testing version](http://tests.icaria.de/MifareClassicTool-2.0.4-testing3.apk). Please check if reading, writing, and the Tag Info tool is working.

jlanza, 2015-07-06:
@ikarus23 it seems it is working correctly now. Could you please post the diff code? ;)

ikarus23, 2015-07-06:
Arrrrrg! But thanks for testing :)
Was reading a tag working with one of the previous testing versions?

jlanza, 2015-07-06:
Unfortunately no. We got an exception. It is quite strange as the Tag
instance seems to be ok. Then we authenticate and it returns true. But
if you get into the functions, the authentication fails.

And therefore, afterwards reading is crashing with and IOException
indicating the tag was lost.

Hope that helps. We can try to work something out, but we are "as lost
as you probably are". It is just the d**n Z3. Do you know which chipset
it includes?

ikarus23, 2015-07-06:
As far as I know it is a NXP chip. A [picture from iFixit](https://d3nevzfk7ii3be.cloudfront.net/igi/R6AEL3D1bg1fiFTr) and [another bug report](https://bugzilla.mozilla.org/show_bug.cgi?id=1136512) indicate this too.

jlanza, 2015-07-06:
So you have the same info I do ;) Therefore I don't know why the hell it
is not able to read Mifare Classic in a proper way. Actually I bought
the phone because of its chipset.

Do you know of a way of sending low level chipset messages within
Android? I would like to do the same way we can do with libnfc.

moscowneversleeping, 2015-07-07:
@ikarus23 ...
ikarus23 commented 3 days ago
I've created another testing version. Please check if reading, writing, and the Tag Info tool is working.

Last version work with all type Mifare Classic and i can see Tag Info too.
thanks

ikarus23, 2015-07-07:
Oooh, so it's working for your tag, moscowneversleeping, but not for the tag of jlanza.

@jlanza What type of tag are you trying to read?

moscowneversleeping, 2015-07-07:
@jlanza 
May be you have Z3 with out NXP NFC chip.
Application Tag Info about NXP can show ALL information Tag.

And what do you used Android version? 
I tested MCT with Android 5.*

jlanza, 2015-07-07:
I'm trying to read a Dual Card. It is a JavaCard with Mifare Classic
emulated. Therefore SAK is 0x38.

When I'm using native Mifare Classic cards, there is no problem either
the app or the mobile phone, ant the card is read properply.
So the problem is not your app, the problem is something with the Z3
software or hardware. As you pointed out before it is really strange
that the techs returned are doubled, on is for Mifare Classic and the
other is linked to the contactless Javacard.

My guess for a solution is, is there any way of attaching a tag to an
specific technology. I mean, when you get NfcA Tag I guess it is the
first one it is found on the list. I wil have to sort something out.

Thanks. keep in touch.

ikarus23, 2015-07-07:
@jlanza (OT)
>Do you know of a way of sending low level chipset messages within Android? I would like to do the same way we can do with libnfc.

As far as I know, there is no easy way to do this. Most phones with NXP chips use the nfc-pn544 kernel module provided by Google/NXP which will create a device at `/dev/pn544` However, I'm not sure how to go from there. Also there is no support for the PN544 by libnfc, as far as I know.

I read a lot of forum threads or mailinglist posts of people who want to have full control over the NFC chip inside the Android phone (e.g. to crack Mifare Classic tag). But I never read of one who succeeded in doing so. However, there has been "research" in that area. I found a document of Hendri Appelmelk, who was able to port libnfc to Android ([original post](https://appelmelk.wordpress.com/2011/02/17/porting-libnfc-to-android/), [pdf mirror](http://tests.icaria.de/libnfc_to_android.pdf)).

Also you may consider asking Michael Roland (e.g. [on stackoverflow]()). This guy really knows his way around NFC and Android!

jlanza, 2015-07-07:
Thanks a lot. I will read the information you attached ;) and if we find
something useful will get back to you.

moscowneversleeping, 2015-07-07:
@jlanza It is Very interesting, may be we can emulate Mifare Classic Tags..?
And may be we can change UID on rewritable tags used special backdoor command?

jlanza, 2015-07-08:
When I meant emulated I mean that the card is a JavaCard with
MifareClassic on the same chipset. This is dual card and there are
several manufacturers that provide this type of cards (Optelio
Contactless, etc.)

[2 more replies]

closed 2015-09-11 by ikarus23