ikarus23/MifareClassicTool #65: Cannot use in Jiayu S3

DickyT, 2015-06-17:
Hi I own a Chinese brand phone Jiayu S3 NFC version. Tech detail here
http://forum.xda-developers.com/android/general/jiayu-s3-5-5-inch-1080p-hd-4g-lte-t3015809

and when i was reading my tag with 100% correct keys, i got 'no valid key found in chosen key file' and i cannot continuing reading my card. Is MCT not yet supported on MTK phones?
Looking for solution, thank you very much.

BTW my os is Android 5.1 with latest MCT

moscowneversleeping, 2015-06-19:
@DickyT Do you try use MCT on your device with Android 4.*?

ikarus23, 2015-06-20:
Hi, do you know what kind of NFC chip is inside this phone? Also, did you try other apps like [NFC Taginfo](https://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo&hl=en), [NFC TagInfo by NXP](https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en) or [NFC Tools](https://play.google.com/store/apps/details?id=com.wakdev.wdnfc)

DickyT, 2015-06-23:
Hi, I flashed back to 4.4 official version yesterday, and I found MCT also cannot use. And I tried NFC TagInfo by NXP, it can read my Mifare Classic 1K card as "MIFARE Classic (MF1S50)" with IC manufacturer is NXP Semiconductors, i think my phone have a fully support on Mifare Classic Cards. And I also tried NFC Tools Pro and NFC TagInfo both can read my card UID, but NFC TagInfo also cannot decrypt my card and show HEX using provided Keys.

Thank you~

DickyT, 2015-06-23:
P.S. The NXP TagInfo also cannot read my Tag HEX data using given Keys.

ikarus23, 2015-06-23:
Hi, I'm sorry but it seams like the phone does not support Mifare Classic.
I will add it to the list of unsupported devices.

DickyT, 2015-06-26:
Hi there, I did some research about this issue, and I found may be this issue is caused by the MTK customized Android, because there is one App call ALIPAY, it can read the Authorized Bus Pass and shows the transaction history and the credit inside. I tried my cards, on of them is Sony Felica and other is Mifare 1K Classic.

ikarus23, 2015-07-14:
This is weird. So you are able to read a Mifare Classic 1k tag? But only with the ALIPAY App?

jekkos, 2015-08-20:
It seems that extra suppport for mifare classic might be on it's way (in AOSP)

https://android-review.googlesource.com/#/c/163641/

DickyT, 2015-08-20:
Hi there, I found some important informations here. In June, my phone
reseller and the customer agent in that company said to me that Jiayu S3
uses NXP chips, but today I found while I entering the Factory Mode of this
phone, and I found the chip inside this MTK phone is MT6605, which is by
MTK. Although I did not found lots of informations of this chip, but i
think I can say that is chip do not fully support Mifare M1 cards.

ikarus23, 2015-09-11:
Added to the list of unsupported devices in e7e70a7fab96190ada81c171b6f839b9844ca429.

closed 2015-09-11 by ikarus23