ikarus23/MifareClassicTool #70: the nfc tag being repeatedly read notification

dfrankes, 2015-08-18:
Hello,

Im 100% Sure my LG G3 can read the tag, But when i want to make the Key Mapping it keeps saying after like 10~20 seconds: The nfc tag being repeatedly read. Reading it repeatedly may consume battery more quickly, and then it loses connection to the NFC Tag

ikarus23, 2015-08-25:
@dfrankes 
This sounds really weird...
Unfortunately I don't own a LG G3 to reproduce this behaviour. Does it occur all the time or only for some tags? What Android version does this device use? What MCT version did you use? Have you tried older ones?

@all
Is there someone else with a LG G3 who can verify this behaviour?

mishel24, 2015-08-29:
I keep getting this error too.  Only thing I have no idea what an NFC tag is???  Sorry to hijack your issue but what does this even mean?  I have the same phone.

dfrankes, 2015-08-29:
Hello,

I used then newest Version of MCT, i didnt try older onces, But im running Android 5.2

ikarus23, 2015-08-29:
@dfrankes Thanks for the info. Could you try to read this tag with other apps? E.g. [NFC TagInfo](https://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo&hl=en) or [NFC TagInfo by NXP](https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en)?

@mishel24 Mhhmm... maybe you take some time and goolge it. ;) I'm sure you know the technology. These credit card shaped plastic cards with a chip inside that can talk with readers wirelessly in close range (e.g. access card for your hotel room). Sometimes in other form factors like key fobs or even stickers. In some cases you can read their memory with your smartphone.

ikarus23, 2016-05-29:
@dfrankes Any news? Have you tried the other apps?

dfrankes, 2016-05-30:
@ikarus23 Noone of the apps work. Didnt test it since.

ikarus23, 2016-05-31:
Puh, so at least I didn't f**ked up. ;)
As @mishel24 is/was getting this error too, I think there is/was something wrong within the Android version that is/was shipped for these devices. If the LG G3 has a new Android version by now, I would really appreciate it if you can test for this behavior once more.

dfrankes, 2016-07-24:
@ikarus23 I will test it later today.

ikarus23, 2017-10-30:
Any news on that?

dfrankes, 2017-11-03:
@ikarus23, Sorry i don't own the LG G3 anymore, Already upgraded to the LG G6

ikarus23, 2017-11-03:
Thanks for your feedback. I say we let this one slide. If someone has the same error in the future I will reopen this issue and try to find more information.

Closed.

unrecognizedguy, 2018-08-08:
Hi! I have the same issue on LG G6

unrecognizedguy, 2018-08-08:
I am running Android version: 8.0.0
Mifare Classic Tool v. 2.2.3

unrecognizedguy, 2018-08-08:
The problem is that Developer option was enabled. If you disable it, the promt disapears

unrecognizedguy, 2018-08-09:
Problem is still occurring

ikarus23, 2018-08-10:
> The problem is that Developer option was enabled. If you disable it, the promt disapears

This sounds really weird. Could you try to read this tag with other apps? E.g. [NFC TagInfo](https://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo&hl=en) or [NFC TagInfo by NXP](https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en) and see if the issue is the same?

groissi, 2018-10-04:
> The problem is that Developer option was enabled. If you disable it, the promt disapears

We also have an App which communicates a bit longer via NFC and we're running into the same issue with that notification message and the NFC connection being reset by the OS of the LG G6.

I retested it with Developer Options enabled and disabled... makes no difference. So **can not** confirm that!

g5user, 2018-12-29:
Hi,
I also have the issue with info "NFC repeatedly read ..... to save battery etc."
I have LG G5 and it was working in March. Now when I try to read tag I have this info and connection is lost. I think it is connected to latest system update but I can't be sure.
Now I have Android 8.0.0 version.
Any hints .... ? 

ikarus23 please help. :-)

ikarus23, 2019-01-15:
I don't think there is anything I can do. It really sound like an issue of LG G3/5/6 devices. I will reopen this issue in hope somebody has more information.

ikarus23, 2019-01-15:
Have you tried to reach out to the LG support?

g5user, 2019-01-16:
I didn't because I think it is connected to Android. I tried to read the same tag on LG G3 and it was perfectly fine. But G3 doesn't have Android 8.

dfrankes, 2019-01-16:
> I didn't because I think it is connected to Android. I tried to read the same tag on LG G3 and it was perfectly fine. But G3 doesn't have Android 8.

I found my old LG G6, i will try the app tomorrow to see if it works (android 7)

ikarus23, 2019-01-17:
> I didn't because I think it is connected to Android.

Well, LG is responsible for the Android version they ship with their devices. There are a lot of customizations manufactures do to their Android. In most cases it's drivers and UI changes. Maybe it connected to the NFC driver they use. But I'm just guessing...

CainSakura, 2019-01-30:
I have an LG V30 running LG stock Android 8.0.0 and facing the same problem.

No matter what tool used (tested with MCT, MTools and other), even without any NFC tool, if a card is connected to the phone for more than 1 minute, then the system will complain about "NFC repeatedly read" and auto disconnect NFC connection.

I also checked that complaining dialog, its activity name is `com.android.nfc.utils.LNfcDialogActivity` within a package named NFC Service `com.android.nfc`.

So I tried to disable the activity in 2 ways (you just pick one you like):
1. Disable this activity using 3C Toolbox (or other similar tools like Lucky Patcher, using `pm disable` method.)
2. Or Just run this command in adb shell, or Terminal Emulator, with root permission.
`pm disable com.android.nfc/.utils.LNfcDialogActivity`

and NFC never complains nor auto disconnect again.

CainSakura, 2019-01-31:
@dfrankes @unrecognizedguy @g5user Please check my solution above.

I use [this app](https://play.google.com/store/apps/details?id=com.willme.topactivity) to check that activity name and package name.

If that command doesn't work, then you probably need to check that activity name by yourself.

dfrankes, 2019-02-01:
@CainSakura that solution seems to be working, but it isn't a good solution. it should work fine without adb shell commands

CainSakura, 2019-02-01:
> how can I do it without root permissions?

@g5user I just searched on the web and seems no way to achieve that.

LeoDupont, 2019-08-22:
LG G6 here, same problem when the reading/writing of a tag is too long.

I first tried with **"Linux's libNFC-only compatible" tags** bought on Amazon (didn't read the description thoroughly before buying), and the writing process was always long enough to trigger LG's NFC protection.

When I realized my mistake, I bought **"Android's MTC-only compatible" tags** and, more often than not, the writing process is fast enough to **not** trigger the NFC protection.  Although I have to try several times for some tags.
So there's that: try several times and check the compatibility of your tags, it might work.

Comensal667, 2020-04-18:
The same problem here usin LG g7 Thinq in Oreo 8.0.0