ikarus23/MifareClassicTool #72: Issues with Xperia Z after updating to Android 5.+

ikarus23, 2015-10-05:
Taken from http://www.proxmark.org/forum/viewtopic.php?pid=18146#p18146:
>I used your great tool many times on my Sony Xperia Z but, after the lollipop upgrade,  the mct mifare classic tool doesn't work with Mifare Mini ATQA 00 04 SAK 9

>i would like inform you that it worked properly on Android 4.4.4 KitKat.
I used your sw to program the TAG on the screenshot.
How can be fixed the issue on new lollipop Android version 5.1.1?

ikarus23, 2015-10-05:
I guess this is somehow related to https://github.com/ikarus23/MifareClassicTool/issues/52 and https://github.com/ikarus23/MifareClassicTool/issues/64 ...

ikarus23, 2015-10-06:
FYI: http://www.proxmark.org/forum/viewtopic.php?pid=18222#p18222

bildin, 2016-02-17:
@ikarus23 I've made beta version of TagDiag application for diagnostics NFC issues on Android
https://github.com/bildin/TagDiag
It allows to view about all accessible information (and rather more with root access)
It collects Info about system build, nfc h/w device and used libs, system class constants and features, activities adjusted for work with nfc.
Also it proxies real tag to log conversation with other apps to find errors.
You can view log in overlay and share it via e-mail.
App works as service. It is controled with Navigation Icon or actions from notification.
On some devices it will proxy as soon as has been started, on some it needs to force tag capture (long press on icon or (+) action).
Short press to dispatch buffered tag again.
Of course it applies patches to proxied tag.
I hope it eill help. Waiting for test reports.

ikarus23, 2016-02-18:
You did it again, bildin! This is a super handy app when it comes to the NFC issues we're expecting lately! Thank you!  The app worked perfectly on my Nexus 7 (2012) / Android 5.1.1.

Hopefully we can nail this issue down using your app.

ikarus23, 2016-05-29:
If you read this, mariolino, have you tried bildins app? Can you provide me with some log?

ikarus23, 2017-10-30:
Lacking feedback. Closed.

closed 2017-10-30 by ikarus23