ikarus23/MifareClassicTool #74: Error: Loosing connection to card before read

timmyjane, 2015-10-15:
HTC M8 Android 5.0.1.  Can not read card.  Software looses connection to card during mapping process.

ikarus23, 2015-10-16:
Have you tried the "auto reconnect" option from the settings menu?
For some tags (with bad antennas?!) it may help...

timmyjane, 2015-10-16:
Yes. Does not help. No issues on my old Galxay II. After the issue showed I tried other NFC readers with the same outcome. The phone looses connection to the tag when trying to map. There must be some type of compatibility issue. Anyone have suggestions?

ikarus23, 2015-11-01:
You tried "other NFC readers"? What do you mead by that? You mean other Android apps? Or other devices (like the ACR122u or other mobiles)?

timmyjane, 2015-11-01:
Sorry. Apps, specifically NFC TagInfo.

ikarus23, 2015-11-01:
Hmmm. It seams like this issue is somehow specific to your mobile or your tag. But since reading the tag worked fine using a Galaxy II, I think it is your device.  The strange part is that many people use MCT with the HTC One M8 device (and Android 5.0.1). So maybe it is the tag after all. Maybe it has such a bad antenna that the strong reader field of the Galaxy II was able to power it, but the HTC One is not. Could you try to read some different Mifare Classic tags (especially with different form factors) with that HTC One device?

timmyjane, 2015-11-01:
Tried on a key fob and still get the same issue. Losing connection.

ikarus23, 2015-11-01:
Do you have a credit card shaped tag near you? In most cases they have much better antennas...

timmyjane, 2015-11-01:
Yes. CC tag is the style that originally looses connection. I have tested 3 different CC style cards with the same result.

ikarus23, 2015-11-01:
Well, that is weird. :(
If other Android apps behave the same way, maybe it is your device after all. Never heard of another HTC One M8 user with the same issues. If you know someone with a HTC One M8 device, you can ask him to read/write your tag to find out if your device is "broken".

timmyjane, 2015-11-01:
Yes very odd. I can only assume that my phone is a POS. Thanks for trying and a great app.

ikarus23, 2015-11-01:
Thank you. You are welcome. I will close this issue now. If you manage to find someone with and HTC One M8 and the same problems, feel free to reopen it.

closed 2015-11-01 by ikarus23