ikarus23/MifareClassicTool #75: Write Tag (Write Block) problems

portmankim, 2015-10-23:
This App is cool, I can use it to read some Mifare 1K tags I have here.
I'm using a new Huawei P8 Lite mobile phone, works fine...except:
- I can read all data from Tags, fast and precise
- I can write data to a Tag via Write Dump (Clone), by selecting 1 Sector a time
- but I can't write data using Write Tag -> Write Block:
- * when using Write Block the App says: "Data Succesfully Written" ..but nothing
- * when I read back the Tag no data has changed
- * the Tags are empty (from manufacturer, all keys are FFFF...)
- * as said before, via Write Dump the data does change on the Tags!!!

What am I doing wrong?
Kim

ikarus23, 2015-11-01:
Hi Kim,

this is really strange. What Android version is in use? Do you have another NFC enabled phone to check if there is the same issue?

psg69, 2015-12-25:
Same problem here.
Samsung Galaxy S5...

psg69, 2015-12-25:
And when it does write via "Write Dump" it is stuffing up and destroying access to all blank sectors...

nestornino, 2015-12-29:
Hi, I'm currently developing an application to write NFC tags, but have a problem when I want to make writing with keys. Reviewing the code Available Project, found in WriteTag.java something I do not understand. The WriteBlock operation (lines 485 and 492), it has arguments that do not match those of the android developers page, which are an integer and a byte just:
http://developer.android.com/intl/es/reference/android/nfc/tech/MifareClassic.html#

I am new to these issues, but I would help her answer.
Thank you

ikarus23, 2015-12-30:
Hi, the `writeBlock()` you see is not the function from the [Android API](https://developer.android.com/intl/es/reference/android/nfc/tech/MifareClassic.html#writeBlock%28int,%20byte[]%29). It is the `writeBlock()` of the [MCReader](https://github.com/ikarus23/MifareClassicTool/blob/master/Mifare%20Classic%20Tool/app/src/main/java/de/syss/MifareClassicTool/MCReader.java#L400) (part of MCT). I hope this helps

ikarus23, 2016-10-09:
Any news on that?

nestornino, 2016-10-09:
Hello, I discovered that the problem was caused because the documentation for Android developers said that for writing to each block, the numbering of each sector was restarted, and I was trying to write in block manufacturer, so it was the error.

I finished the application, but I put aside, because I want to establish a business model through sale tag in my country.

Thanks for asking about my project.

ikarus23, 2017-10-30:
Lacking feedback of OP. Closed.

closed 2017-10-30 by ikarus23