ikarus23/MifareClassicTool #77: Latest build crashes on open CM13 Nexus 5X

drsn0w, 2015-12-02:
Will update issue with relevant logs tomorrow afternoon. Th previous release works fine, the latest one force closes as soon as the app is open

ikarus23, 2015-12-02:
Did it work before?
Other users report it also crashes on the Nexus 6P...

dnet, 2015-12-17:
I have the same issue Nexus 5X, but with Android 6.0.1. Here are the relevant lines of log:

```
12-17 10:35:34.947  4780  6758 I ActivityManager: Start proc 11297:de.syss.MifareClassicTool/u0a92 for activity de.syss.MifareClassicTool/.Activities.MainMenu
12-17 10:35:34.999 11297 11297 W System  : ClassLoader referenced unknown path: /data/app/de.syss.MifareClassicTool-1/lib/arm64
12-17 10:35:35.059 11297 11297 E MainMenu: Error while creating '/MifareClassicTool/key-files' directory.
12-17 10:35:35.063 11297 11297 D AndroidRuntime: Shutting down VM
12-17 10:35:35.063 11297 11297 E AndroidRuntime: FATAL EXCEPTION: main
12-17 10:35:35.063 11297 11297 E AndroidRuntime: Process: de.syss.MifareClassicTool, PID: 11297
12-17 10:35:35.063 11297 11297 E AndroidRuntime: java.lang.RuntimeException: Unable to resume activity {de.syss.MifareClassicTool/de.syss.MifareClassicTool.Activities.MainMenu}: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.app.AlertDialog.hide()' on a null object reference
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3103)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.app.ActivityThread.handleResumeActivity(ActivityThread.java:3134)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2481)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.app.ActivityThread.-wrap11(ActivityThread.java)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:102)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:148)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:5417)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
12-17 10:35:35.063 11297 11297 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.app.AlertDialog.hide()' on a null object reference
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at de.syss.MifareClassicTool.Activities.MainMenu.checkNfc(MainMenu.java:452)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at de.syss.MifareClassicTool.Activities.MainMenu.onResume(MainMenu.java:415)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.app.Instrumentation.callActivityOnResume(Instrumentation.java:1258)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.app.Activity.performResume(Activity.java:6327)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        at android.app.ActivityThread.performResumeActivity(ActivityThread.java:3092)
12-17 10:35:35.063 11297 11297 E AndroidRuntime:        ... 10 more
12-17 10:35:53.699 11297 11297 I Process : Sending signal. PID: 11297 SIG: 9
12-17 10:35:53.740  4780  6863 I ActivityManager: Process de.syss.MifareClassicTool (pid 11297) has died
```

dnet, 2015-12-17:
It seems https://github.com/ikarus23/MifareClassicTool/blob/master/Mifare%20Classic%20Tool/app/src/main/java/de/syss/MifareClassicTool/Activities/MainMenu.java#L452 references `mEnableNfc` which is `null` at the time.

ikarus23, 2015-12-17:
Hey there,
thanks for the stack-trace. No that I see it, it looks kind of familiar. I think I stumbled upon this NullPointerException before. I will look into it and fix it.

ikarus23, 2015-12-17:
Would you be so kind to check out [testing version of MFC](http://tests.icaria.de/MifareClassicTool-2.0.5-testing.apk) I just build?

dnet, 2015-12-17:
Thanks for the build, it started at least, but I don't have a Mifare Classic with me right now. I'll test it tomorrow, but at least the crash at startup seems to be solved.

dnet, 2015-12-18:
It discovers the tag, and the `Tools > Tag Info` activity shows the UID/ATQA/SAK values correctly (same response as from Proxmark3). However `Read tag` doesn't do anything -- the screen just flicks and the main activity reloads (or sometimes, just the button press animation shows, but nothing else happens).

Relevant lines from log:

```
12-18 14:10:38.083  4903  6950 I ActivityManager: START u0 {act=android.nfc.action.TECH_DISCOVERED flg=0x20000000 cmp=de.syss.MifareClassicTool/.Activities.MainMenu (has extras)} from uid 10092 on display 0
12-18 14:10:38.084  4903  6950 W ActivityManager: startActivity called from non-Activity context; forcing Intent.FLAG_ACTIVITY_NEW_TASK for: Intent { act=android.nfc.action.TECH_DISCOVERED flg=0x20000000 cmp=de.syss.MifareClassicTool/.Activities.MainMenu (has extras) }
12-18 14:10:38.206  6907 21196 E NxpExtns: Error Sending msg to Extension Thread
```

(the last line repeats several times)

```
12-18 14:10:39.440  4903  6489 I ActivityManager: START u0 {cmp=de.syss.MifareClassicTool/.Activities.ReadTag} from uid 10092 on display 0
12-18 14:10:39.477  4903  6489 I ActivityManager: START u0 {cmp=de.syss.MifareClassicTool/.Activities.KeyMapCreator (has extras)} from uid 10092 on display 0
12-18 14:10:39.558  6907 21196 E NxpExtns: Error Sending msg to Extension Thread
```

(again, last line repeated several times)

The only lines I deleted were all about the audio subsystem playing some sound effect. This device has an NXP NFC chipset in it, and the [official NFC TagInfo app](https://play.google.com/store/apps/details?id=com.nxp.taginfolite) can read the contents of the card without problem, so the scope of the bug is probably within the app itself.

KRLY05, 2016-01-22:
Tried on my nexus 5x stock android 6.0.1:
Reading is ok, but while writing i receive 3 different errors randomly:
- Some error occurred while writing.
- Tag lost while checking for keys with wright privileges.
- Some parts are not writable! Check the whole list below.
After few tries I received "Data successfully written" and it was ok

ikarus23, 2016-01-22:
Well, this is really odd. Unfortunately I don't own a Nexus 5x/6p or any Android 6.x device. I will keep an eye out. Maybe one of my friends got one. I think it would be a lot easier for me to debug this with access to such a device.

ikarus23, 2016-01-29:
Maybe [this](https://github.com/ikarus23/MifareClassicTool/pull/80) is of any help.
(But I think it's not. At least for KRLY05s issues it's not.)

moscowneversleeping, 2016-01-30:
@ikarus23 
Hi! Do you remember me? I wrote some time ago about Z3 and Android 5.*
https://github.com/ikarus23/MifareClassicTool/issues/64

And now I have Nesus 6P H1512 model
I tested version from https://github.com/ikarus23/MifareClassicTool/pull/80 2.0.6-testing.
It is stable. I can readand write tags, Thank you.

moscowneversleeping, 2016-01-31:
@ikarus23 programm work, but not so good
As i write in #64 on Nesus 6p i can not read some classic tags too
[image]

ikarus23, 2016-02-01:
Hmm...  there is something really strange with some of your Mifare Classic tags. The ATS is uncommonly long. My Mifare Classic tag have no ATS. This is a emulated Mifare Classic tag, right?

However, this is kind of *off topic* since it is not related to the crash issues on the Nexus 5x...

ikarus23, 2016-02-05:
The crash issue on the Nexus 5x should be fixed within the latest release (version 2.0.7).

closed 2016-02-05 by ikarus23

f4cket, 2016-06-28:
I have a nexus 5X and have this problem. If anyone solved this please tell me how. /k3ff

f4cket, 2016-06-28:
Im on 2.0.7

f4cket, 2016-06-28:
Have tried different custom roms, cm13 dirty unicorns and pure nexus with and without nethunter kernel. Jolla kernel as primary kernel(multirom)
Pure nexus is android 6.0.1

ikarus23, 2016-06-28:
The crash I knew was right at the startup of MCT. But as far as I remember it, it was due to the permissions update with Android 6.0+. Also, I thought I had it tested...

Can you describe the error in more detail? Or is it just at the startup?