ikarus23/MifareClassicTool #78: Marshmallow permissions (with suggestion)

systemcrash, 2016-01-21:
It will probably need something like this in MainMenu.java

    private static final int REQUEST_WRITE_STORAGE = 112;

    @Override
    public void onRequestPermissionsResult( int requestCode, @NonNull String[] permissions, @NonNull int[] grantResults) {
        super.onRequestPermissionsResult(requestCode, permissions, grantResults);
        if (requestCode == REQUEST_WRITE_STORAGE) {
            if (grantResults[0] == PackageManager.PERMISSION_GRANTED) {
                //permission granted  start reading
            } else {
                Toast.makeText(this, "No permission to read external storage.", Toast.LENGTH_SHORT).show();
            }
        }
    }

        if (ActivityCompat.checkSelfPermission(MainMenu.this, Manifest.permission.READ_EXTERNAL_STORAGE) !=
                PackageManager.PERMISSION_GRANTED) {
            ActivityCompat.requestPermissions(MainMenu.this, new String[]{Manifest.permission.READ_EXTERNAL_STORAGE},REQUEST_WRITE_STORAGE);
            return;
        }

See
http://stackoverflow.com/questions/33139754/android-6-0-marshmallow-cannot-write-to-sd-card

I got MFCT to start on MM after compiling the above additions in, although it crashed once around:

mEnableNfc.hide();

ikarus23, 2016-01-23:
Hi systemcrash, thanks for the code snipped. I will look into it as soon as I got my hands on a Android 6.x device. ;)

Also the crash around `mEnableNfc.hide();` should be fixed within the latest release (2.0.6).

ikarus23, 2016-01-29:
Have a look at https://github.com/ikarus23/MifareClassicTool/pull/80.

ikarus23, 2016-02-04:
MCT should work on Android 6.x. now (https://github.com/ikarus23/MifareClassicTool/pull/80 was merged).

closed 2016-02-04 by ikarus23

systemcrash, 2016-02-12:
Confirmed - works on my MM device.