ikarus23/MifareClassicTool #82: Dont write 0 sector on China card

aleks921, 2016-02-10:
I try to change 0 sector in China card with changable uid. But get error. But other sectors soft write sucessfull.

My scrennshot:
http://hostingkartinok.com/show-image.php?id=eac3be3a6b216afaa137a4a20ce3a247

Please, help me. How write mifare classic changable card uid in 0 sector ?

ikarus23, 2016-02-11:
There are two types of UID changeable tags:

1.  Block 0 is writable with a simple write command.
2.  UID is changeable with special commands.

Unfortunately, it is not possible to perform the special commands with Android. Therefore you can only use the first (1.) kind of tags with MifareClassicTool. It is most likely you got yourself the wrong type of tag.

You can change the UID of type 2 tag using a common USB RFID reader (such as the ACR122u) or special tools like the Proxmark3.

aleks921, 2016-02-11:
I want to buy a card that is a simple command is written in block 0, and where these can be taken and how many they cost ? Can you give a link to aliekspress?

ikarus23, 2016-02-11:
On AliExpress there are a lot of tags that claim to be "block 0 direct writable". However not all of them are. Quite a lot are only writable with special commands, although they say they are "directly writable". There is no way to be 100% sure. Check the comments section! In may cases you will find users who already tested the tags for the "direct write" ability.

There are reliable sources too. But they are much more expensive. For example check out [this website](http://www.clonemykey.com/mifare-classic-tool-mct-compatible/). Usually this company also sells the tags via ebay, but lately the shop seems to be down.

Good luck with purchasing a "block 0 direct write" tag!

aleks921, 2016-02-12:
I want to buy mifare ultralight with change UID. These cards do not have such problems how mifare classic with special commands ? Where they can be on aliekspress or ebay to buy cheap?
Can you give link to aliekspress on Ultralight card or Ultralight coin ? 

P.S. Do you speak Russian ? I am from Russia.

ikarus23, 2016-02-13:
Sorry, I have no experience with UID changeable Mifare Ultralight tags. Maybe one of the guys in the [Proxmark3 forum](http://proxmark.org/forum/index.php) can help you out. And no, I can't speak Russian.

closed 2016-02-15 by ikarus23