ikarus23/MifareClassicTool #85: I write wrong BCC byte

aleks921, 2016-05-13:
I write in classic 1k with direct uid changable BCC byte incorrect.
Help me, please, how restore this card ?
With an invalid BCC byte this card does not see my phone with nfc and acr122u.
How restore, please.

ikarus23, 2016-05-15:
I've never done this. But I think it is not possible to fix cards with a wrong BCC. Have a look at https://github.com/ikarus23/MifareClassicTool/issues/81.

aleks921, 2016-05-15:
Card with special comand i fixed after bcc byte) How fixed with direct comand i don't know.

ikarus23, 2016-05-27:
Added in 57ff1a60f43a14130fcf8bc78e8b6c08c4d77258.

closed 2016-05-27 by ikarus23

aleks921, 2016-05-27:
How update MCT for me on smartfon. Download from google play or not.

ikarus23, 2016-05-28:
Yes, I will release the new version of MCT on Google Play within the next days.

aleks921, 2016-05-28:
Ok, what is number will in new version. Wait new version:)