ikarus23/MifareClassicTool #86: read block0

akingyin1987, 2016-05-16:
some times i read block0  

The same label read out the data is not the same
amsung J5 mobile phone

I am from Chinese, a little English

ikarus23, 2016-05-19:
Hi there, I'm not sure if I understand your issue. You read out block 0 multiple times and some times the read out data is different?

akingyin1987, 2016-05-19:
yes,This problem has occurred in the Samsung J5 and HUAWEI C199

The same Tag, you have been reading occasionally appear, but I only found these two phones

ikarus23, 2016-05-19:
So maybe you got a tag that is emulating a Mifare Classic tag. Some of these change the UID every time you read them.

akingyin1987, 2016-05-20:
NO ，That is the problem of mobile phones, HUAWEI mobile phone upgrade did not find this problem

ikarus23, 2016-05-20:
So the Samsung J5 reads out random data for block 0 and for the Huawei works well with MCT? I have no idea where this behaviour came from...

ikarus23, 2016-10-09:
Is there there anyone out there that can reproduce this behavior? Is the Samsung J5 known for having issues with MIFARE Classic?

systemcrash, 2016-12-19:
Your "libnfc" checks should correctly detect whether a device supports MFC via NFC. Some devices might have faulty blobs/stacks which need updating or fixing.

ikarus23, 2017-10-30:
Lacking feedback. Closed.

closed 2017-10-30 by ikarus23