ikarus23/MifareClassicTool #87: Help needed: Samsungs Galaxy S6 and S6 Edge

ikarus23, 2016-05-26:
Hi everyone,

could those of you with a Samsung Galaxy S6 or a Galaxy S6 Edge do me a favor?
Could you install the latest version of MCT from [here](http://publications.icaria.de/mct/releases/2.0/) and tell me if you are able
to read and write tags? I'm still not sure if those devices are capable of reading/writing
Mifare Classic tags.

Thanks!

Cheers
  ikarus

Th3ju, 2016-06-02:
Hi Ikarus,

Just try with the latest version on my galaxy S6, but still unable to read or write a mifare classic 1K tag :-(

Cheers,
Juju

ikarus23, 2016-06-03:
Thank you! So it seams like the NFC controller in the Galaxy S6 does not support Mifare Classic tags.

I wonder if this is also true for an Galaxy S6 Edge... lakobosj from [this thread](http://forum.xda-developers.com/galaxy-s6-edge/help/nfc-chip-s6-t3081471). said it is possible to read Mifare Classic tags with the S6 Edge, but not to write them. I never heard of such behavior. I think what he meant was that he can read the UID.

ikarus23, 2016-06-12:
lasersword wrote in the [proxmark3 forum](http://www.proxmark.org/forum/viewtopic.php?pid=22286#p22286):
> It runs on s6 edge pperfectly

I will take of the Samsung Galaxy S6 Edge of the incompatibility list and make MCT available through Google Play again.

closed 2016-06-12 by ikarus23

fr3nsis, 2016-12-07:
https://github.com/ikarus23/MifareClassicTool/issues/107

Zeldaelite, 2017-09-08:
I'm having the same issues.