ikarus23/MifareClassicTool #89: htc one m8 don't read mifare mini tags after uodate to 5.0.1 lolipop

fakal, 2016-06-15:
htc one m8 don't read mifare mini tags after update to 5.0.1 lolipop... before that android version works fine.

best regards

ikarus23, 2016-06-19:
Hi fakal! Thanks for reporting this issue. It looks like HTC has f**ked up another Android build.
This is not the first time... have a look at https://github.com/ikarus23/MifareClassicTool/issues/52. You can try [@bildin's app](https://github.com/bildin/TagDiag/raw/master/bin/eu.dedb.nfc.tagdiag.apk) (from https://github.com/ikarus23/MifareClassicTool/issues/72) to look into what is
going wrong.

Ok, but maybe it is a issue of MCT after all... What error have you encountered? Does the app crash
during reading/writing? or while mapping the keys?

fakal, 2016-06-20:
it gives a screen with yellow exclamation mark and "said This device does not support Mifare Classic"

and bellow says generic info
can read UID, RF tech, ATQA SAK
only in ATS says "-"

ikarus23, 2016-06-28:
From this information it is really hard to tell what is wrong. As I said, you can try [@bildin's app](https://github.com/bildin/TagDiag/raw/master/bin/eu.dedb.nfc.tagdiag.apk) (from https://github.com/ikarus23/MifareClassicTool/issues/72) to look into it. There should be an option to create log file. You can post it here or send it via mail.

Before that you should try other NFC apps like [NFC TagInfo](https://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo), [NFC TagInfo by NXP](https://play.google.com/store/apps/details?id=com.nxp.taginfolite) or [NFC Tools](https://play.google.com/store/apps/details?id=com.wakdev.wdnfc&hl=en). If they work I did something wrong. If not, it is most likely that HTC did something wrong.

ikarus23, 2016-10-09:
Any news on this? Have you tried other apps?

fakal, 2016-10-10:
i byed another phone, but yesterday the app self updated , and i check the s6 egde and the htc m8 none can read the mifare mini, and still doesn't work with other apps...

ikarus23, 2016-10-10:
Sorry to hear that. Looks like the issue is within those phones and their Android ROMs.

ikarus23, 2017-10-30:
Closed because the issue is most likely an error by HTC.

closed 2017-10-30 by ikarus23