ikarus23/MifareClassicTool #91: How to encrypt one sector ?

Pascal66, 2016-06-25:
What's the way to encrypt one sector (15 for example) with only KeyA on a fully formated mifare 4K ?
For testing purpose

ikarus23, 2016-06-28:
I think there is some misunderstanding how Mifare Classic works. Mifare Classic does not encrypt any sector with KeyA or KeyB. Those keys are for authentication purposes and access conditions only. Have a look at Access Condition Decoder/Encoder tool of MCT and of course, the [Mifare Classic datasheet](http://www.nxp.com/documents/data_sheet/MF1S50YYX.pdf) (section 8.7).

Pascal66, 2016-06-29:
I nuderstand that, my fault, i've not explained that  (dont know how) I wish a way to change AC more clearly.
I've already changed the sector trailer (it work), but the answer doesnt show the same than another card.
Ie my own sector trailler show -----------------  AC  FFFFFFFFFFFFF  and the other show nothing than "cant read sector".

ikarus23, 2016-07-04:
I'm not sure what you did and I'm not sure if I understand what you want to do. If you want to change the keys and the Access Condition just write to the Sector Trailer. But you should know what you are doing. You can destroy a sector by writing faulty ACs. Please have a loot at the datasheet (as I said within my last post)! You will find all information about ACs and Sector Trailers in there.

Pascal66, 2016-07-05:
From: ikarus 
Sent: Monday, July 4, 2016 6:56 PM
To: ikarus23/MifareClassicTool 
Cc: Pascal ; Author 
Subject: Re: [ikarus23/MifareClassicTool] How to encrypt one sector ? (#91)

I'm not sure what you did and I'm not sure if I understand what you want to do. If you want to change the keys and the Access Condition just write to the Sector Trailer. But you should know what you are doing. You can destroy a sector by writing faulty ACs. Please have a loot at the datasheet (as I said within my last post)! You will find all information about ACs and Sector Trailers in there.

Pascal66, 2016-07-05:
Hello,
I read the datasheet every day lol.
I want to know from a card : 
- the actual AC (I cant read it with mct)
- knowing if only KeyA or KeyB or both are used to protect a sector (the 15 for my side)
I know that my question are newbie styled, but i'm trying to ask with clear words

ikarus23, 2016-07-05:
* If you know KeyA (in some cases KeyB will work too) you can read the ACs of a sector. If not, it is most likely that your tag is broken.  You can't read the ACs without any key.
* Just decode the ACs with MCT's "Access Condition Decoder & Encoder" (in the tools section). It will show you which key is needed to perform an action on a block.

Pascal66, 2016-07-05:
Does that mean i need to bruteforce (or mfoc mfcuk) a least KeyA to access of the AC ?

From: ikarus 
Sent: Tuesday, July 5, 2016 7:50 PM
To: ikarus23/MifareClassicTool 
Cc: Pascal ; Author 
Subject: Re: [ikarus23/MifareClassicTool] How to encrypt one sector ? (#91)

  a.. If you know KeyA (in some cases KeyB will work too) you can read the ACs of a sector. If not, it is most likely that your tag is broken. You can't read the ACs without any key. 
  b.. Just decode the ACs with MCT's "Access Condition Decoder & Encoder" (in the tools section). It will show you which key is needed to perform an action on a block.

ikarus23, 2016-07-05:
Yes it does. The ACs are always readable with KeyA (and sometimes with KeyB too).

closed 2016-07-19 by ikarus23