ikarus23/MifareClassicTool #92: Works well in OnePlus3! The new mobilephone 2016

darf2k, 2016-07-04:
Hi,ikarus23:
I just to report that MCT works well in OnePlus3! The new mobilephone 2016.

Happy as I?

HAHA, except...the "special commands" for the changeable UID card from China...still...un~~

Bytheway, for a long time I would like to know:"Do you know exectlly what does the `special commands` look like? And the 'simple write command'?" T____T    Hunger to know.

ikarus23, 2016-07-04:
Hi darf2k, thanks for the good news!

About the commands: You can look up the *normal* Mifare Classic commands from its [datasheet](http://www.nxp.com/documents/data_sheet/MF1S50YYX.pdf). Have a look at section 9.1 on page 15. The special commands can be found somewhere in the Proxmark3 forum. Have a look at the [posts were I tried to implement the *special* commands in MCT](http://www.proxmark.org/forum/viewtopic.php?pid=7015#p7015).

darf2k, 2016-07-06:
Thanks for reply~
I now study hard to understand.
Learning it all throug last night, and now I am at daylight working which is totally deffrent from coding programms. In China, you can see.
And I found these:
**"from：wan chang
at：https://www.zhihu.com/question/36918910/answer/86410655

The special commands are:
Sent bits: 26 (7 bits) //to find card  0x26 / 0x52 
Received bits: 04 00
Sent bits: 93 20 //prevent from crash
Received bits: 01 23 45 67 00
Sent bits: 93 70 01 23 45 67 00 d0 6f //select card
Received bits: 08 b6 dd （SAK）
（DONNT CONFIRM ,OR CANNT CHEAT）

Sent bits: 50 00 57 cd //halt，50 00 就是 PcdHalt()
Sent bits: 40 (7 bits) （special command）//the first special command for Backdoor，to set BitFramingReg to sent 7 bits
eceived bits: a (4 bits)
Sent bits: 43 （special command）//the seconde special command
Received bits: 0a
Sent bits: a0 00 5f b1 //normal first handshake
Received bits: 0a
Sent bits: 00 dc 44 20 b8 08 04 00 46 59 25 58 49 10 23 02 c0 10 //write block 0 
Received bits: 0a

THE IMPORTANT THING IS：
1. DONT MAKE 3 Pass Authenticaiton
2. SENT PcdHalt()
3. SENT 0x40 in 7-bit
4. SENT 0x43"**

Just like what you`ve told me.
I`m learning.....
thank U.

ikarus23, 2016-07-06:
Thanks for the nice summery on the special (backdoor) commands. What are you trying to achieve? Do you just want to understand how the backdoored tags work? Or do you want to write an Android app that can change the UID of backdoored tags?

alfs, 2016-07-06:
Sending the special commands is not possible from standard android since the first command is using a non-checksummed code. See for example https://github.com/Proxmark/proxmark3/blob/83f11ccf729655f17c92b4cf9ae4e10278f8b2e3/armsrc/mifarecmd.c#L1090

Android's [trancieve()](https://developer.android.com/reference/android/nfc/tech/NfcA.html#transceive(byte[])) function, that is normally used to send such commands, adds the checksum bytes itself. You may be able to circumvent this with a custom firmware however and have a trancieve-function where you can specify exactly what bytes to transmit.

ikarus23, 2016-07-07:
Exactly. This Is why I asked him if he wanted to achieve this. ;-)

darf2k, 2016-07-14:
Hi~ ikarus23 und alfs, thank U for sharing informatios above!
I'm indeed a fresh bird in testing NFC tec, and only driven by vagary. 
What I wantto do is:
First,  to completely copy a M1card by using a android phone.
Second, to find out how to simulate a M1 card by using also a same moblephone( seems that depends on whicht type the NFC chip inside is).
And always 3rd 4th5th ideals come will…
All your words are so useful for me, thanks again.

BTW, mates, do U know that there is a very new type of China magic nfc card is available? Which can be rewritten UID  in 00 block BY THE VERY ORIGINAL EDITION MCT APK HERE, and they are called"CUID CRAD". Of course I've verified before told U. I think it uses the "normal command" to do these, by hardware redesigning.
Want to get some? I can mail U some of those cards!  Tell me where to mail by writting to darf@163.com.  ^_^)Y

Happy hacking and making~

ikarus23, 2016-07-15:
Hi darf2k. The thinks you want to do will not be easy. Especially the emulating with a mobile phone. You will have to hack deep into the android ROM and the driver (kernel module) of the NFC chip. I never saw a working example of an Android device emulating a Mifare Classic tag. So good luck!

Yes, I know of these newer tags. Thats why a added the feature to MCT to change block 0 of those tags. But it is not that easy to get them here in Europe from a reliable  source. I only have one of this kind of tag. Maybe I will write you a mail to get another ;) Thank you for the offer anyway!

ikarus23, 2016-07-19:
Closed.
(No issue, just a OT discussion.)

closed 2016-07-19 by ikarus23