ikarus23/MifareClassicTool #95: 2K card support

bildin, 2016-08-05:
Hi!
I've seen it in your TODO list.
Could you add this feature?
May be start just with checkbox in settings, that allows trying to read sectors up to 32?

ikarus23, 2016-08-07:
Hi @bildin,
Thanks for reminding me of my TODO list :) I'm really not that active on this project lately...

Just for the context: You want to read a Mifare Plus tag, right? Because I don't know any Mifare Classic tag that has 2k memory. Mifare Plus configured in Security Level 1 (SL1) should be fully compatible to Mifare Classic, if I'm not mistaken.

Cheers!

PS: Is Android detecting your tag as 1k Mifare Classic, although it's a 2k Mifare Pluss?

bildin, 2016-08-08:
Yes, I am talking about Mifare Plus in SL1 mode.

In Android there is no difference between 1K and 2K card detecting.

Relying to [AN10833 (MIFARE Type Identification Procedure)] (http://cache.nxp.com/documents/application_note/AN10833.pdf) document there are no correlated flags to exclusively determine 2K cards:

[image]
[image]

Just only if SAK equals 0x10 (Mifare Plus SL2), but this case is parsed by Android well.

However, [android sources](http://grepcode.com/file/repository.grepcode.com/java/ext/com.google.android/android/5.1.1_r1/android/nfc/tech/MifareClassic.java#MifareClassic) tell:
>     private static void validateSector(int sector) {
>         // Do not be too strict on upper bounds checking, since some cards
>         // have more addressable memory than they report. For example,
>         // MIFARE Plus 2k cards will appear as MIFARE Classic 1k cards when in
>         // MIFARE Classic compatibility mode.
>         // Note that issuing a command to an out-of-bounds block is safe - the
>         // tag should report error causing IOException. This validation is a
>         // helper to guard against obvious programming mistakes.
>         if (sector < 0 || sector >= MAX_SECTOR_COUNT) {
>             throw new IndexOutOfBoundsException("sector out of bounds: " + sector);
>         }
>     }

At the same time [NFC TagInfo by NXP] (https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=ru) determines 2K cards.

Using my [TagDiag] (https://github.com/bildin/TagDiag) I see some extra commands before reading the tag:

```
TRANSACTION_connect (2) from com.nxp.taginfolite
Connect MIFARE_CLASSIC (8) result SUCCESS (0)
TRANSACTION_isPresent (6) from com.nxp.taginfolite
isPresent: true
TRANSACTION_resetTimeouts (16) from com.nxp.taginfolite
TRANSACTION_reconnect (3) from com.nxp.taginfolite
Reconnect result SUCCESS (0)
TRANSACTION_connect (2) from com.nxp.taginfolite
Connect MIFARE_CLASSIC (8) result SUCCESS (0)
TRANSACTION_transceive (7) from com.nxp.taginfolite
> E080
< 0C75778002C1052F2F01BCD600
TRANSACTION_transceive (7) from com.nxp.taginfolite
> C2
< Transceive failed.
TRANSACTION_isPresent (6) from com.nxp.taginfolite
isPresent: true
TRANSACTION_transceive (7) from com.nxp.taginfolite
> 614482A849804B791BEA7BCC !
(Mifare auth block 0x44 with KEY B: 4B791BEA7BCC)
< Transceive failed.
TRANSACTION_transceive (7) from com.nxp.taginfolite
> 62
< B2
TRANSACTION_resetTimeouts (16) from com.nxp.taginfolite
TRANSACTION_reconnect (3) from com.nxp.taginfolite
Reconnect result SUCCESS (0)
TRANSACTION_connect (2) from com.nxp.taginfolite
Connect MIFARE_CLASSIC (8) result SUCCESS (0)
```
Then starts usual iteration over sectors with known keys. If tag is Mifare Plus iteration continues up to block 0x7C (sector 31), otherwise (in case of 1K card) - up to block 0x3C (sector 15).

Cheers!

bildin, 2016-08-08:
So I can assume:
E080 - command to identify Mifare Plus;
6144... - Authentication to block beyond 1K bounds to check memory availability;

If you need, I can send a card sample via mail.

ikarus23, 2016-08-16:
Hi @bildin! Sorry for not answering earlier and thank you for all the information. I will try to implement the "fixed sector count" feature as soon as possible. However this will not be that soon... For the next two weeks I will be on vacation.

About that card... I really don't have any Mifare Plus tag for testing. Could you write me a email? You will find a address in the git commit log or on the Google Play page of MCT.

Another off-topic question: Do you know your way around BCC? On 4 byte UID tags, it is the 5th byte of the very first block. But do you know if there is some thing like this for 7 or 10 byte UID tags? (Execpt for the communication. The BCC will always be part of the UID read process.) Maybe you could have a look at [this thread](http://www.proxmark.org/forum/viewtopic.php?id=2274).

ikarus23, 2016-10-04:
MIFARE 2k support has been merged into the master. It will be part of the next release.

closed 2016-10-04 by ikarus23