ikarus23/MifareClassicTool #97: add disable autostart option when card detected

treakiandroid, 2016-09-05:
hi,
could you please add an option to disable autostart when a card is detected? 

thanks

ikarus23, 2016-09-05:
As far as I know, this is not possible. Have a look at https://github.com/ikarus23/MifareClassicTool/issues/71

treakiandroid, 2016-09-23:
it should be possible because it is working with mensaguthaben ( https://github.com/jakobwenzel/MensaGuthaben ) maybe you can adapt there code to add an option to disable autostart. thanks

ikarus23, 2016-10-03:
Hmm, I tried the app, but it did not work for me. Have you tested this successfully?

ikarus23, 2017-10-30:
Any news on that? Is there anyone that knows a way to do this?

ikarus23, 2018-09-10:
Lack of feedback. Closed.

closed 2018-09-10 by ikarus23

ikarus23, 2019-03-31:
It works since d6eb7d150cc186509718c18abcfe1ba9ddc8f6b2 thanks to @aerizzo. The feature will be included in the next release.