ikarus23/MifareClassicTool #98: HTC 10 , Android 6.0.1

dimitrire, 2016-09-26:
Hello,  mifare classic tool doesn't work on Android 6.0.1 , what do you think please ?

dimitrire, 2016-09-26:
the application says it is not a Mifare card while it worked fine on my htc one m7 with android 5.2

ikarus23, 2016-09-27:
This is odd again. There is a lot of strange behavior for HTC devices with Mifare Classic and Android 6+. (e.g. see https://github.com/ikarus23/MifareClassicTool/issues/89)

[This site](http://www.shopnfc.it/en/content/7-nfc-device-compatibility#htc) is claiming that the HTC 10 is not capable of reading/writing Mifare Classic tags. So MCT saying "this is not a Mifare Classic tag" might be a false positive...

Have you tried other NFC apps like [NFC TagInfo](https://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo), [NFC TagInfo by NXP](https://play.google.com/store/apps/details?id=com.nxp.taginfolite) or [NFC Tools](https://play.google.com/store/apps/details?id=com.wakdev.wdnfc)?

dimitrire, 2016-10-07:
Hello I tested other applications but just detect the card, I made mistakes screens printed MCT, more when I read the map, areas that can only be read are random, that have never the same .

dimitrire, 2016-10-07:
[image]
[image]
[image]
[image]
[image]

ikarus23, 2016-10-07:
Ok, this is really strange. I never had a device that is working sometimes and sometimes not. Some of this issues might be caused by a bad antenna in the tag or the phone. but I'm not sure. Do you have other tags?

Have you ever read sector 0 successfully? On figure 3 it looks like this sector might be broken. Or this is some other false positive... And what is the content of the tag you are trying to write? Could you paste it here?

dimitrire, 2016-10-07:
Sector 0
74C21961CE880400C232000000000012
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
Sector 1 to 7 and 15
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
Sector8
20150C5E2876A56F8A35415003E10B07
1DC6A24E73A030CCD5B0479FC002883C
BE1270FBFB9952277A14F1BBFD5E4AD4
A0A1A2A3A4A50F00FFFF415A54454B4D
Sector9
6561AC1FC1A4E47B4959E7F1B0257470
CC8001659650A596BA87FF8F364B527D
C3BB2E514BC5311C9043BE710C549483
A0A1A2A3A4A50F00FFFF415A54454B4D
Sector10
BA87FF8F364B527D875EBCFDC50D3D28
4BCA79F4567E28A14959E7F1B0257470
BB0DDD075D499BE6BA87FF8F364B527D
A0A1A2A3A4A50F00FFFF415A54454B4D
Sector11
C3BB2E514BC5311C9043BE710C549483
BA87FF8F364B527D6867DE3057B907DB
D6A2FAA164E1B3B9BA87FF8F364B527D
A0A1A2A3A4A50F00FFFF415A54454B4D
Sector12.14
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
A0A1A2A3A4A50F00FFFF415A54454B4D
Sector13
D7FD3B612A8E04D2BA87FF8F364B527D
00000000000000000000000000000000
00000000000000000000000000000000
A0A1A2A3A4A50F00FFFF415A54454B4D

dimitrire, 2016-10-07:
The tag works on the machine

dimitrire, 2016-10-07:
The dump is to the HTC one m7

dimitrire, 2016-10-25:
The mifare was writed one shot of ten on the HTC 10

ikarus23, 2017-01-08:
It really looks like another Android 6+ NFC issue. Maybe this is not a fault of HTC. However, it is not an issue within MCT. Maybe it is related to [this Android bug](https://code.google.com/p/android/issues/detail?id=228344).

ikarus23, 2017-10-30:
Well, it really looks like exactly the same issue I'm having with my Google Nexus 5X. I'm closing this as it seams like it is a duplicate of https://github.com/ikarus23/MifareClassicTool/issues/106 and because it is most likely an Android issue...

closed 2017-10-30 by ikarus23