ikarus23/MifareClassicTool #99: Zenphone 2 z00a

bruno-oliveira1, 2016-09-29:
Hi your APP read tag information, but show this message tour device (due to hardware) dowles not support Mifare classic. How to fix it ? Thank you very much

ikarus23, 2016-10-03:
Hi! Thanks for reporting that the Zenphone 2 z00a is not compatible with MCT.

As you can read in the error message, there is nothing we can do. It seams like this phone has a NFC controller that is not manufactured by NXP and is therefore not able to read/write MIFARE Classic tags.

bruno-oliveira1, 2016-10-03:
You know if a buy any one nxp RFID reader I can read Mifare Classic tags ? If not why you recommend ? Thank you very much again.

ikarus23, 2016-10-03:
The ACR122u RFID reader is a very common to read/write Mifare Classic tags. But there are much cheaper ones.

closed 2016-10-09 by ikarus23