ikarus23/MifareClassicTool #100: A very strange coffee dump, hexa encrypted I am in the blur

hannibalxking, 2016-10-08:
hi everyone its all in the tittle i not understated what its going on whit the dump  for my coffee machine.
in my original card i have 11.45€ but inside the dump nothing no result of course for hex a 479
its intriguing i want to know what the problem or i need to buy another eyes *.*
the dump is here if you are curious and god help you friend
[coffee11.45.txt](https://github.com/ikarus23/MifareClassicTool/files/517351/coffee11.45.txt)

ikarus23, 2016-10-09:
It looks like the blocks are encrypted. However, some blocks have the exact same content... I think you need to buy another coffee and dump the tag again. Look for changes (e.g. with MCT's diff function). You can also try to ask the guys at the [proxmark forum](http://www.proxmark.org/forum/index.php) for help (and provide additional information, e.g. what coffee machine are we talking about).

hannibalxking, 2016-10-09:
hi thank you for you response i already have another dump with 10.0€ where take a look the change its big for just for a money less   
[coffee 10€.txt](https://github.com/ikarus23/MifareClassicTool/files/518483/coffee.10.txt)
the machine its a french machine called mend's no wifi or nothing weird 

ps ; MCT work great with the meizu mx4 pro

ikarus23, 2016-10-29:
Sorry, but I can't tell from those dumps what exactly is going on there. However, a simple hack you could try is just to restore the 10€ dump once the card is down to 5€ or something like this.

I will close this question because it is a bit off topic (regarding the MCT development). However, feel free to comment on your progress with these coffee machine tags.

closed 2016-10-29 by ikarus23