ikarus23/MifareClassicTool #101: Save Key Map

steets250, 2016-10-11:
I know that this issue was discussed in issue #43, but it was closed last year. 

Is there a plan to implement a method to save key maps so that MIFARE cards can be read more quickly and efficiently? I scan the same card quite often and having to wait for it to create the key map each time I want to read the card seems like there should be a better way.

Thanks!

ikarus23, 2016-10-11:
#43 was just closed by accident and has been reopend short after.

I know, for reading the same tag over and over again, MCT sucks. But introducing key maps is a major change and until now I do not had the time to do it. This might be also true for the future. I'm just not sure when I will manage to implement this (same like #13).

ikarus23, 2016-10-15:
Closed. Still covered by #43

closed 2016-10-15 by ikarus23