ikarus23/MifareClassicTool #103: App data storage

osysltd, 2016-10-26:
Dear Ikarus,

First of all thank you for such high quality development!

A quick question\feature request:
Is it feasible to store the app data in the /data partition (or add an option to store the info there) as this is the only place which is being encrypted on the encryption-enabled smartphone. The aim is to be on the safe side having custom dictionary keys and dumps and prevent unauthorized access to the information.

Thank you.

ikarus23, 2016-10-28:
Hi osys,
this seams like a reasonable option. I put the data on the external memory so everybody can
easily backup or modify them and they do not get lost if the app gets uninstalled. However, if
you store real secret keys then of course it would be best to store them in the /data partition.
I will add an option that allows you to switch between internal and external memory. But as
always, I'm not sure when I will have the spare time to implement it. ;)

ikarus23, 2017-03-05:
I implemented this feature. You can try it using [this MCT testing version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing.apk). There will be a release containing this feature once I had a look into some other construction sites of MCT.

bxl0608, 2017-03-05:
Thanks very much! I will install it.

时间：周日 3月5日 22:55
收件人：ikarus23/MifareClassicTool <MifareClassicTool@noreply.github.com>
抄送：Subscribed <subscribed@noreply.github.com>

osysltd, 2017-03-05:
Unfortunately I cannot install on Android 5.1.1.
*** Previous version from GooglePlay uninstalled

ikarus23, 2017-03-05:
Yes, you cannot have both versions installed at the same time. You have to uninstall MCT before you install this version. Your dumps and key files will survive this. However, if you use the "save on internal storage" option from this version, all data in the internal storage will be gone once you uninstall it again. Have a look at the information on the new option, which are available within the app.

ikarus23, 2017-03-05:
Maybe I will add an import/export function for files from and to the internal storage. But I'm not sure if that's worth the effort. If the keys in your key file are that secret, you probably should not carry them on your phone all the time...

osysltd, 2017-03-05:
Previous version from GooglePlay was uninstalled prior to the installation.

osysltd, 2017-03-10:
Dear Ikarus,
Thank you for your reply. I've deleted any mct from device and installed the latest version. In the settings set up the option to use the internal storage, however can see that the storage folder is appearing in two places: /storage/emulated/0/MifareClassicTool and also /data/data/de.syss/MifareClassicTool. Is that expected? Default key files are equal in these places

ikarus23, 2017-03-10:
Hi @osysltd, yes this is expected. MCT creates the basic structure at both places. I know that only data at /data/data/... is protected. As long as you use the new option new key and dump files will be stored at /data/data. You can switch when ever you want.

osysltd, 2017-03-10:
Cool! thank you very much for the efforts, now mct users can be on the safe side.

ikarus23, 2017-03-10:
You are welcome. Thank you for testing!

closed 2017-03-10 by osysltd