ikarus23/MifareClassicTool #106: Strange Bug on Nexus 5X (Dev Needs Help)

ikarus23, 2016-11-20:
Hi MCT users,

I recently bought a Nexus 5X to continue the MCT development on a newer platform (my Nexus 7 2012 has Android 5.1.1 and there will be no updates). However, I experience a really strange behavior and was not able to figure out what is causing it.

* I have clean, factory formatted tag (all key are FFFFFFFFFFFF).
* If I read it with my Nexus 7 2012, everything is just fine.
* If I read it with my Nexus 5X, some sectors appear to be not readable.
* If I read it with my Nexus 5X again, some different sectors appear to be not readable. The readable sectors vary from read to read.

As far as I have looked into it, the two reasons for a unreadable sector change for every read approach.
* Sometimes the sector is not readable  because no key was found (the authentication fails even with the correct key).
* Sometimes it is not possible to read the sector (even though the key is correct).

**To all Nexus 5X users:**
Do you experience the same issue? (unreliable read results) I have them regardless of what app I'm using (MCT, NFC TagInfo, NXP's Taginfo, etc.)

eastwoodnet, 2016-11-21:
My Nexus 5X (H798 with 32G )  has same issue.
the same key files can't work on it ,but on Nexus 7 (2012)  it's ok!

ikarus23, 2016-11-21:
Hmm. That's what I thought. So maybe it is an issue with the Android ROM itself. I will report this to Google and see what happens.

ikarus23, 2016-11-21:
https://code.google.com/p/android/issues/detail?id=228344

eastwoodnet, 2016-11-28:
BTW, My Nexus 5X is running android 7.0, but on Android 6.0.1,it still can't work. I use stock rom not root it.

sasodoma, 2017-06-29:
Not sure if this is still relevant but my 5X reads the tags perfectly almost every time. Make sure the card is on the reader (behind the camera) and that it doesn't move. I am running 7.1.2 but I know it worked on 7.0 too.

ikarus23, 2017-12-27:
I'm not sure, but if feels like it is better now that I'm on Android 8.1

ikarus23, 2018-01-04:
Can some of the Nexus 5X/6P users confirm this?

anowlcalledjosh, 2018-01-04:
It doesn't really seem any better to me, I just tried to read the same tag several times and it never gave a stable result – sometimes a whole sector was marked as bad wrongly, sometimes it failed to read one or more of the keys (but got the rest of the sector)...

ikarus23, 2018-01-05:
Hmmm. Maybe my results were better just because I used another tag. Maybe I just got lucky. Maybe this issue is connected to antenna problems as well...

bennesp, 2018-01-11:
Nexus 5X, Android 8.1, always had these problems in both reading and writing. I had to do any operation many times before it goes well

I always had this issue, since android 5 as far as I remmeber, and always accused the unreliable Nexus 5X NFC chip, but I had never known if It was true

ikarus23, 2018-01-11:
Well, I found out why it worked for me better. I had the "retry authentication" option enabled by accident. With this option MCT retries an authentication if it fails once. Since it's the authentication that is causing the issues on the Nexus 5X it will work 50% better with the option enabled.

Maybe, as a workaround, I should modify the option to something like "retry authentication X times". With a great X the error should be minimized. However, reading time on a tag with not all keys known will increase a lot.

sasodoma, 2018-01-11:
Now that other users mention it I guess my last post is not accurate. Ever since I first tried MCT back on Marshmallow, the tags were never fully read. About 50% of the time there would be a sector missing and i had to repeat the scan in order to get the full dump. And I believe it has never read both of the keys in every sector but I just thought that's how it works since it didn't need both keys to read. When I wanted a full dump including keys I would just save 3 scans and combine the missing keys.

bennesp, 2018-01-11:
As @ikarus23 said, enabling "retry authentication" helps a lot.
~95% of the readings were performed successfully

ikarus23, 2018-01-11:
@bennesp: Thanks for the feedback.

@sasodoma: Wow, no, that is not normal! If you have all the keys, a normal Mifare Classic tag and a device without any NFC issues (e.g. Nexus 7 2012) everything should work first try without missing stuff.

anowlcalledjosh, 2018-01-11:
Yep, "retry authentication" definitely helps. Without it almost every read would have at least one key or sector missing, but every read I did with the option enabled worked fine.

sasodoma, 2018-01-11:
I can also confirm that the retry setting eliminates missing sector. And it also reads almost all of the keys. Sometimes it still doesn't find both keys though.
The way I see it the app just tries every key in the file against every sector and if it authenticates then it saves it. Now before it would fail on a lot of keys so some block didn't have any keys and couldn't be read. Now it just finds like 90% of the keys so every block has at least one key to read it. Do I understand the behavior right?

ikarus23, 2018-01-11:
@sasodoma: Yes, MCT just does a so called dictionary attack on the tag with the keys from the file(s). And with "retry authentication" it tries each key twice if it fails. The Nexus 5X sometimes reports that the authentication failed, although it was the right key. If this happens twice in a row, the sector or the key will marked as unknown.

ikarus23, 2018-01-12:
I implemented the "retry authentication" option with a custom count. If I set it to something like "10" on my Nexus 5X the dump looks fine.

It would be great if you guys could test it and play around with the number of retries. You can download the testing version [here](http://tests.icaria.de/MifareClassicTool-2.2.1-testing1.apk).

ikarus23, 2018-01-12:
Thanks for testing. So to be sure to not miss anything a value of 2+ seems to be needed.

ravem, 2018-01-15:
I have set the number of retries to 10 on my N5x and, so far the problem seems solved (both reading and writing a dump).

micolous, 2018-02-08:
I have had a lot of trouble getting the Nexus 5X authenticating reliably with some MFC tags. Retrying authentication many times helps a lot.  I had to add retry logic to my own application to make it work well, and I default to 5 retries.  It does make the whole process extremely slow, especially when users don't have the keys...

I suspect it's a design issue with the phone rather than a software bug, sometimes the communication with the tag is too weak, especially with older tags.  This is true even when the phone is not in a case.  I've seen it on some other phones as well, but I can't recall what model, because they weren't mine.

ikarus23, 2020-08-16:
Since this issue was never fixed on Googles behalf and I implemented a workaround which helps a bit, there is nothing more to do here. Also, Nexus 5X user base is getting smaller...

I will close this issue for now.

closed 2020-08-16 by ikarus23