ikarus23/MifareClassicTool #107: samsung s6 edge g925f not supported

fr3nsis, 2016-12-07:
with latest updates samsung has removed again the compatibility with MIFARE Classic tags?

ikarus23, 2016-12-17:
Thank you for reporting this. I will add it to the list of incompatible devices.
I think I heard that MCT is working well on the S6 Edge. Was there an Android update?

ikarus23, 2017-01-08:
Stupid question. Of course the was an Android update. You already said so. Maybe this broken MIFARE Classic support is because of an Android 7+ update (see: https://code.google.com/p/android/issues/detail?id=228344). However, I will add the S6 Edge to the list of incompatible devices again.

osysltd, 2017-01-09:
Dear Ikarus,
Speaking about compatible devices, Yota phone (yd206, Chinese version) works fine.

ikarus23, 2017-01-23:
TagDiag output of an Samsung Galaxy S6 Edge:
```#FIRMWARE INFO
Build:
BOARD = universal7420;
BOOTLOADER = G925FXXS4DPJ2;
BRAND = samsung;
CPU_ABI = arm64-v8a;
CPU_ABI2 = ;
DEVICE = zerolte;
DISPLAY = MMB29K.G925FXXS4DPJ2;
FINGERPRINT =
samsung/zeroltexx/zerolte:6.0.1/MMB29K/G925FXXS4DPJ2:user/release-keys;
FOTA_INFO = 1475741828;
HARDWARE = samsungexynos7420;
HOST = SWDD6303;
ID = MMB29K;
MANUFACTURER = samsung;
MODEL = SM-G925F;
PRODUCT = zeroltexx;
RADIO = unknown;
SERIAL = 02160228f350b9a6;
TAGS = release-keys;
TYPE = user;
UNKNOWN = unknown;
USER = dpi;
getRadioVersion() = G925FXXS4DPIJ;
VERSION.BASE_OS =
samsung/zeroltexx/zerolte:6.0.1/MMB29K/G925FXXU4DPGW:user/release-keys;
VERSION.CODENAME = REL;
VERSION.INCREMENTAL = G925FXXS4DPJ2;
VERSION.RELEASE = 6.0.1;
VERSION.SDK = 23;
VERSION.SECURITY_PATCH = 2016-10-01;

#HARDWARE INFO
NFC devices:
/dev/sec-nfc
NFC libs:
/system/lib/libnfc-nci.so
/system/lib/libnfc_nci_jni.so
NFC configs:
/system/etc/libnfc-sec-hal.conf
/system/etc/libnfc-sec.conf

#SYSTEM INFO
INfcTag supported transactions:
TRANSACTION_close = 1;
TRANSACTION_connect = 2;
TRANSACTION_reconnect = 3;
TRANSACTION_getTechList = 4;
TRANSACTION_isNdef = 5;
TRANSACTION_isPresent = 6;
TRANSACTION_transceive = 7;
TRANSACTION_ndefRead = 8;
TRANSACTION_ndefWrite = 9;
TRANSACTION_ndefMakeReadOnly = 10;
TRANSACTION_ndefIsWritable = 11;
TRANSACTION_formatNdef = 12;
TRANSACTION_rediscover = 13;
TRANSACTION_setTimeout = 14;
TRANSACTION_getTimeout = 15;
TRANSACTION_resetTimeouts = 16;
TRANSACTION_canMakeReadOnly = 17;
TRANSACTION_getMaxTransceiveLength = 18;
TRANSACTION_getExtendedLengthApdusSupported = 19;
TagTechnology supported techs:
NFC_A = 1;
NFC_B = 2;
ISO_DEP = 3;
NFC_F = 4;
NFC_V = 5;
NDEF = 6;
NDEF_FORMATABLE = 7;
MIFARE_CLASSIC = 8;
MIFARE_ULTRALIGHT = 9;
NFC_BARCODE = 10;
android.hardware.nfc: true
android.hardware.nfc.hce: true
com.nxp.mifare: false

#ACTIVITIES INFO
TECH_DISCOVERED:
com.sec.android.app.shealth/com.samsung.android.app.shealth.sensor.sdk.accessory.background.NfcEventActivity
b4a.example/b4a.example.main
com.android.apps.tag/com.android.apps.tag.TagViewer
de.syss.MifareClassicTool/de.syss.MifareClassicTool.Activities.MainMenu
TAG_DISCOVERED:
HOST_APDU_SERVICE:

#LOG:
State changed: NFC ENABLED
State changed: NO TAG
State changed: TAG IS PRESENT
Discovered TAG:
4 byte UID: 212448E4
1 supported technologies:
1 android.nfc.tech.NfcA
Tech extras:
short sak = 0x08
byte[] atqa = 0400
---
nativeHandle: 1
INfcTag: android.os.BinderProxy@51643f1

Proxied TAG:
4 byte UID: 212448E4
1 supported technologies:
1 android.nfc.tech.NfcA
Tech extras:
short sak = 0x08
byte[] atqa = 0400
---
nativeHandle: 1
INfcTag: eu.dedb.nfc.tagdiag.NfcDiagService$NfcProxyBinder@948bd6

Dispatching... (pick activity)
State changed: BUFFERED TAG
State changed: TAG IS PRESENT
Discovered TAG:
4 byte UID: 212448E4
1 supported technologies:
1 android.nfc.tech.NfcA
Tech extras:
short sak = 0x08
byte[] atqa = 0400
---
nativeHandle: 1
INfcTag: android.os.BinderProxy@51643f1

Proxied TAG:
4 byte UID: 212448E4
1 supported technologies:
1 android.nfc.tech.NfcA
Tech extras:
short sak = 0x08
byte[] atqa = 0400
---
nativeHandle: 1
INfcTag: eu.dedb.nfc.tagdiag.NfcDiagService$NfcProxyBinder@948bd6

Dispatching... (pick activity)
State changed: BUFFERED TAG
Dispatching... (pick activity)
Received broadcast eu.dedb.nfc.BROADCAST_VIEW_LOG
Received broadcast eu.dedb.nfc.BROADCAST_VIEW_LOG
Received broadcast eu.dedb.nfc.BROADCAST_SHARE
Received broadcast eu.dedb.nfc.BROADCAST_SHARE
```

ikarus23, 2017-01-23:
I will add the Samsung Galaxy S6 Edge to the list of incompatible devices. At least temporarily...

Stoom, 2017-10-22:
Is this an issue with the Exynos chipset?  On a  Snapdragon chipset of my S8+ it was able to read both my prox card for home and my yubikey neo.

ikarus23, 2017-10-22:
I don't think this as something to do with the application processor (snapdragon). It is either an issue with the NFC chip or just an software/driver bug.

ikarus23, 2017-10-30:
@fr3nsis, any news on this? Have you tried it again lately?

Stoom, 2017-11-19:
@ikarus23  Once I found the keys for the card it looks like it was able to write almost everything to a magic card on an US S8+

ikarus23, 2017-12-27:
Closing. The S6 and S6 Edge is on the list of incompatible devices.

closed 2017-12-27 by ikarus23

Stoom, 2017-12-27:
Sorry, just noticed this... The S8+ US variant can write the payload, but not the magic card manufacturer block.

ikarus23, 2017-12-28:
So the S8+ is now able to write Mifare Classic? Interesting. Maybe I should remove it from the list of incompatible devices.

The fact that you can't write to the manufacturer block is most likely due to your tag type. There are different types of tags which are UID changeable. Only some of them are compatible with Android. Have a look at [general information section](https://github.com/ikarus23/MifareClassicTool#general-information).