ikarus23/MifareClassicTool #111: Question: looking for a new compatible phone from 2016

arrabal, 2016-12-30:
Hey guys, I screwed my HTC One M7 when trying to replace the internal battery. Opening the phone was easy, I didn't even damage the screen, which everyone claimed that could be easily broken during opening. However, i pulled a flex cable too strong and ended breaking a motherboard connector. 

So, I think it's the time to buy a new phone, but I'd like to get a modern one, from 2016, that is compatible with MCT. I've read the INCOMPATIBLE_DEVICES.md file and it seem that the list of compatible phones is very reduced and they're all quite old. 

Is there any new model that is known to work? I'm thinking about Xiaomi Mi 5s or Huawei P9. I'm also thinking about newer HTC models, but I had too many problems with my old M7 (pink camera, loose USB dock, wifi disconnection, internal battery...)

Thank you in advance.

juancols, 2017-02-26:
Hi Arrabal,

I just tried MCT with the Huawei P9 a couple of days ago and it works perfectly. I could read and write a MIFARE Classic 1k key without any issues. By the way, the NFC antenna is right on the upper speaker of the phone instead of the back side.

Regards!

ikarus23, 2017-10-30:
This discussion is old. Closed.

closed 2017-10-30 by ikarus23