ikarus23/MifareClassicTool #117: Feature reguest Automatic Read and dumb

amahlaka, 2017-02-27:
Hi, i attempted to add a feature that allows the app to wait until a tag is present and then read and dumb it to a file, however i got stuck in how to actually make it wait for tag without causing a ANR.
can someone either help me making this feature or can someone make this?

ildar, 2017-02-27:
A wild guess: use non blocking API while waiting?

amahlaka, 2017-02-27:
i tried using Callable, but it never seems to exit the while loop, i tried ` while((reader = MCReader.get(mTag)) == null){` and ' while(mTag == null){' but no luck

ikarus23, 2017-02-27:
I think the right way to do this, is to create an extra Activity. Like some "Quick Dump Tool". This activity should use the foreground dispatch system (you can inherit this from MCT's [BasicActivity](https://github.com/ikarus23/MifareClassicTool/blob/master/Mifare%20Classic%20Tool/app/src/main/java/de/syss/MifareClassicTool/Activities/BasicActivity.java)). You might want to overwrite the `onNewIntent` method. This method is called automatically if (among other things) a new tag was found.

ikarus23, 2017-02-27:
As for you feature request: I think its a duplicate. Check out https://github.com/ikarus23/MifareClassicTool/issues/43. However, this is a really old feature request and I don't think I will have the time to implement it soon.

ikarus23, 2017-03-10:
Closing this issue in favour of https://github.com/ikarus23/MifareClassicTool/issues/43.

closed 2017-03-10 by ikarus23