ikarus23/MifareClassicTool #118: Cant read read tag

LonerDan, 2017-03-06:
Hello,
I can´t map keys to sectors. It tells me that "no valid key was found in chosen key file", but i know for fact that the card uses default key (FFFFFFFFFFFF). I tested the card(s) with PN532 reader under linux and it works fine.
I have Jiayu S3+ with Android 5.1.1, kernel 3.10.65+.

ildar, 2017-03-07:
Faq?

LonerDan, 2017-03-07:
Sorry, what do you mean? I should have also mentioned that the app worked on my phone before, but sadly I dont remember which version it was (It was long time ago).

ikarus23, 2017-03-10:
Hi!
* There are a lot of devices with broken MIFARE Classic support after updating to Android 5+
* Maybe this is connected to https://github.com/ikarus23/MifareClassicTool/issues/66.
* Try to find out which version of MCT worked. You can download all the old versions [here](https://publications.icaria.de/mct/releases/).

osysltd, 2017-03-10:
Also per my tests smartmx with classic emulation are not working well. Maybe thats your case?

ikarus23, 2017-03-11:
If it actually is smartMX ,you might want to try out the MCT option "Auto reconnect". I added it way back because it helped some guy to use MCT with smartMX tags.

osysltd, 2017-03-11:
@ikarus23 thanks for the tip, will check

LonerDan, 2017-03-11:
Thx, I will check the previous versions you provided. Also, the app worked on 5.1.1 before, I didn't upgrade my system. Btw, is there any way to debug the app?

ikarus23, 2017-03-12:
I'm not sure what you mean with debugging, but yes, of course you can debug the app. It's an open source app after all. So just download Android Studio and the app's source code from this repository and you are good to go. Step-by-step debugging, breakpoints, etc. - it's all there.

ikarus23, 2017-10-30:
Lacking feedback. Closed.

closed 2017-10-30 by ikarus23