ikarus23/MifareClassicTool #119: Facing Issue while building the app in Eclipse

rhlkumar09, 2017-03-18:
Hi I am getting the following error

Description	Resource	Path	Location	Type
error: No resource identifier found for attribute 'showAsAction' in package 'android'	dump_editor_functions.xml	/MainMenu/res/menu	line 23	Android AAPT Problem

Description	Resource	Path	Location	Type
error: No resource identifier found for attribute 'showAsAction' in package 'android'	general_options.xml	/MainMenu/res/menu	line 24	Android AAPT Problem

Description	Resource	Path	Location	Type
AndroidManifest.xml file missing!	MainMenu		Unknown	Android Packaging Problem

kindly let me know how to resolve this

rhlkumar09, 2017-03-18:
i have downloaded the zip file from github and create an apk using eclipse IDE the APK crashes if i install it in device

ikarus23, 2017-03-19:
Hi @rhlkumar09, I migrated the MCT repository to the IntelliJ-based  [Android Studio](https://developer.android.com/studio/index.html), which is the official Android development platform for some time now. I'm not sure if it will still be able to build with the old Eclipse based IDE. Maybe you want to give Android Studio a try? ;)

closed 2017-05-26 by ikarus23