ikarus23/MifareClassicTool #122: Write manufactor block after writing dump

maxzaycev, 2017-04-02:
I choose write Manufactor block and write dump. After writing 0 block of 0 sector cards are disconnected. I think that good idea write manufacture block after writing remaining sectors.

ikarus23, 2017-04-02:
There should not be any disconnect after writing the manufacturer block. There is no difference between writhing block 0 and other blocks.

What device are you using? What is the Android version? What kind of tag do you have? Have you ever written the block 0 successfully with MCT? Or with any other device/software? Have you tried the "auto reconnect" option of MCT?

maxzaycev, 2017-04-02:
1. Nexus S (i9023)
2. Android 4.4.4 (CM 11 Stable)
3. Mifare Classic 1K (S50) with supporting Changeable Manufactor block (Aliexpress)
4. Yes, I`m writing block 0 successfully on MCT and My App (On this Nexus S devices). But the tag always disconnected after block 0 writing. 
5. Auto reconnect not solved this problem. As far as I understand, this option using during only key mapping process.

If your cards work correctly, then the problem is exclusively in my cards.

ikarus23, 2017-04-03:
Hmm. The card I have does not disconnect after block 0 writing. You could be right, this might be exclusive for your card. Or it might be an issue with CM 11. You could try Googles stock rom. However, I doubt that this will solve the issue.

bildin, 2017-04-06:
I have the same issue after 0block writing. So I write 0block separately from other dump.
This is normal when UID is changed NFC controller sees other card in the RF field in autopolling subroutine and NFC Service will send a new intent.

ikarus23, 2017-04-06:
Sounds fairly reasonable. Although I never experienced it. Maybe I should movie writing block 0 to the end, as @maxzaycev suggested.

alpha-1983, 2017-10-05:
No MIFARE Classic or Plus card will let you ever write to block 0 of sector 0. That block is known as the manufacturer block, and it's value is fixed during production.
There's a dedicated entry in all of the MFC and MFP datasheets indicating that the block 0 is write protected.
If a MIFARE Classic write command is used to try to write to that block, the card will treat it as a reset. If a ISO14443-4 based write command is used in MIFARE Plus, you'll get an error code.

ikarus23, 2017-10-05:
For original MIFARE Classic tags you are absolutely right. But we were talking about "magic" cards here. There are special MIFARE Classic tags that support writing to the manufacturer block with a simple write command. This App is able to write to such tags and can therefore create fully correct clones. However, some special tags require a special command sequence to put them into the state where writing to the manufacturer block is possible. These tags will not work.

ikarus23, 2017-10-30:
Taken from https://github.com/ikarus23/MifareClassicTool/issues/134
>After  a bit of thinking, I don't think there is a solution to fix the write  dump behaviour for block 0 changeable tags which disconnect themselves  after block 0 was changed.
>
>If I write block 0 last, the key with write privileges might not be  valid anymore or block 0 is no longer writable due to the access  conditions.
>
>If I don't write block 0 last, it might disconnect. Android does not  notice this disconnect. And I can't find any way to force Android into  rediscovering tags.
>
>Therefore, I don't think there is anything I can do to fix this. User  with this issue must write block 0 separately. Maybe I will try to  implement a mechanism that detects this issue and displays a suitable  error message.
>
>Just a side note: I have a block 0 changeable tag that does not  disconnect after writing. So there are tags out there which work  perfectly fine. ;)

stirb520, 2018-01-19:
I want to know why mct cant support some special tags that require a special command? My tags maybe are the special tag, i want to write it。

ikarus23, 2018-01-20:
@stirb520 it is not possible because the Android NFC API does not allow to send the special commands. Some of the issues are mentioned [here](http://www.proxmark.org/forum/viewtopic.php?pid=7175#p7175).