ikarus23/MifareClassicTool #123: Use of MCT together with app External NFC

34victor, 2017-04-07:
When using the MCT together with Eхternal NFC application, the Read and Write Tags buttons are not available. Can this be fixed? Thank you!

ikarus23, 2017-04-07:
Yes, better External NFC support is right on top of my MCT-todo-list. ;)
However, it is likely that I will have not time before the end of May. Please be patient.

ikarus23, 2017-04-07:
Hopefully this will cover the 3 years old issue https://github.com/ikarus23/MifareClassicTool/issues/13 too...

34victor, 2017-04-10:
As they say in Russia - the promised three years are waiting.  ))))

k1r4v, 2017-07-01:
I live in Brazil, I would like to use my ACR122U device with this app... It would be awesome! Most devices here doesn't have NFC hardware, (expensive mobile phones models only have it). This app made my life really easier: there's not similar app on windows or linux available on the market that does what this app do. My latest mobile phone with NFC embedded died and now I am unable to use this app anymore on others mobile phones I have here (I bought donate version).

ikarus23, 2017-07-03:
Sorry to hear that. And sorry for the delay. I'm in contact with @bildin, the developer of [External NFC](https://play.google.com/store/apps/details?id=eu.dedb.nfc.service). We are working on putting a solution together to make MCT work smoothly with External NFC.

If you do not have Android 7.x it might work already... not good and still buggy but it *might* work...

k1r4v, 2017-07-03:
It's good to hear! I'll follow the issue! Your work is great, keep going on!

ikarus23, 2017-11-17:
I finished a beta version of MCT with support for External NFC. On my setup it was still unstable and on NFC enabled devices with an external NFC reader attached you might experience some "strange" behavior. But if you want to give it a try, feel free to do so. http://tests.icaria.de/MifareClassicTool-2.1.0-testing8.apk

ikarus23, 2017-11-17:
As for me: I use a [RFID-RC522](https://www.amazon.com/dp/B016ZBKDLI/) reader and a [PL2303](https://www.amazon.com/dp/B00KM6X7FC/) based serial adapter. I can discover 4-byte-UID-tags just fine. However, when I try to read them, just after the key mapping process, I get an error ("tag lost"). After this the symbol of External NFC just stays at the "wave animation" with waves going from left to right. I'm no longer able to discover tags until I restart the External NFC service. Also, I'm not able to discover a 7-byte-UID-tag.

If you have the same trouble (with the same setup) it might be a hint that it is an issue within External NFC. But not sure yet.

Rrtk2, 2017-11-17:
Hey,
I too have a ACR122U device, is this update also incorporated in MifareClassicTool-2.1.0-testing10.apk? If so i could test the performance of the compatibility ACR122U.

ikarus23, 2017-11-17:
Hi @Rrtk2 and thanks for offering your help again. No, testing8 was build from the [improve_startup_checks](https://github.com/ikarus23/MifareClassicTool/tree/improve_startup_checks) branch and testing10 from the master. I will merge them as soon as I got some feedback from @bildin (developer of External NFC) and have worked out some other minor issues. After this you should be able to test if the ACR122U is working. I'm looking forward to it since I know a lot of people have an ACR122U (but I don't).

ikarus23, 2017-12-17:
Good news everyone. I bought myself a ACR122U and tested MCT with External NFC using this [test build](http://tests.icaria.de/MifareClassicTool-2.1.0-testing8.apk). It worked great! I will merge the changes into the master branch and build a release (probably around 2017-01-07, hopefully earlier).

ikarus23, 2017-12-27:
The code is now in the master branch. Still waiting for feedback from @bildin regarding https://github.com/ikarus23/MifareClassicTool/issues/123#issuecomment-34523793.

ikarus23, 2018-01-04:
This feature has been released with version 2.2.0 (a44921d5156996d0a6ed1d7e8469fe7a5daefb67).

closed 2018-01-04 by ikarus23

ikarus23, 2018-05-11:
Just some more test results of a user using External NFC:
>I have made further attempts, and surprizingly enough, the WRITE feature
does work with the MFRC522 / CP210x "N2elite" USB reader.
So, I can write dump files that I have previously read with an ACR122 and/or
manually edited.

So it still seams like External NFC has some issues.

kgamecarter, 2019-01-20:
Extenal NFC 1.04b version updated at 2019 1/17.
Fixed the crash issue.