ikarus23/MifareClassicTool #125: Samsung Galaxy S7 Edge compatible. Android 7.0

rjsilvestre, 2017-04-10:
Although listed as incompatible and not able to install from Play store I installed the APK and the MCT works perfectly.
Maybe the model could be delisted.

Thanks!

ikarus23, 2017-04-11:
Hi @rjsilvestre, thank you for reporting this. What exact model do you have?

rjsilvestre, 2017-04-11:
@ikarus23 the model is SM-G9350
Tell me if you need any other information.

ikarus23, 2017-04-11:
Thanks, but unfortunately google does not list the device with this name.
There are the following S7 Edge devices available: 
```
SamsungGalaxy S7 Edge– SCV33
SamsungGalaxy S7 Edge– hero2qltecctvzw
SamsungGalaxy S7 Edge– hero2lte
SamsungGalaxy S7 Edge– hero2qltespr
SamsungGalaxy S7 Edge– hero2ltelgt
SamsungGalaxy S7 Edge– hero2lteskt
SamsungGalaxy S7 Edge– hero2qltetmo
SamsungGalaxy S7 Edge– SC-02H
SamsungGalaxy S7 Edge– hero2qltechn
SamsungGalaxy S7 Edge– hero2qlteue
SamsungGalaxy S7 Edge– hero2qlteusc
SamsungGalaxy S7 Edge– hero2qlteatt
SamsungGalaxy S7 Edge– hero2qltevzw
SamsungGalaxy S7 Edge– hero2ltebmc
SamsungGalaxy S7 Edge– hero2ltektt
```
Do you know which one of these could be your device?

rjsilvestre, 2017-04-11:
hero2qltechn
Had to check on AIDA64
Thanks!

ikarus23, 2017-04-12:
Thanks! I will update the list and remove the device from the black list in Google Play.

ikarus23, 2017-05-26:
Updated List too in 9cca1ca807ccaf434e28b05f9179f87f9bc2201f.

closed 2017-05-26 by ikarus23