ikarus23/MifareClassicTool #127: One Plus One compatibility

ArchangeGabriel, 2017-04-21:
Hi there,

Thanks for this very great tool! I was surprised to see it supposed not to be working one the OnePlus One, because it works on mine (but doesn’t work on a friend one though).

Could that be OS dependent? I’m running CyanogenMod 13.0 currently (but might switch to LineageOS or OmniROM at some point), while my friend is running CyanogenOS 12.1 I think.

I’m happy to provide any useful info if that could make it work for other OPO users!

ikarus23, 2017-04-25:
That is indeed interesting. Thank you for reporting. I'm not sure if there are different models or generations of the OnePlus One, but I think you are right: It's just the software. I found a lot of posts online saying something like "NFC is currently only partially supported". Maybe they worked it out. 

It would be great if you can talk to your friend in order to verify that you both have the exact same hardware and only different software (operating system).

ArchangeGabriel, 2017-05-26:
@ikarus23 No, sorry. My friend updated it’s phone from COS 12.1 to Lineage OS, but hasn’t yet answered me on whether this works.

ikarus23, 2017-05-26:
Ok, thanks. So I will wait a little bit longer. An one of my friends has a OnePlus One too. I will ask him if he can confirm this.

ArchangeGabriel, 2017-06-09:
Got news from my friend, it does work for him now that he is on LineageOS. ;)

ikarus23, 2017-06-09:
Thank you. That is most interesting. I will add the device to the list of supported devices with a remark regarding the CM/LineageOS version.

ikarus23, 2017-10-30:
Added in 626498a6dffcaa95091e28ea309b250754811ba5. Closed.

closed 2017-10-30 by ikarus23