ikarus23/MifareClassicTool #129: No valid key found in chosen key file

mark-monserrat, 2017-04-25:
The app works fine at first, I can read/write data on d card. Then all of a sudden when I tried to increment sector 1 block 0, a toast appear saying No valid key found in chosen key file'. I'm sure I have the right keys. From then on I can't access the card anymore. It keeps showing 'No valid key found in chosen key file'.

mark-monserrat, 2017-04-25:
In addition even if I'm using same sets of key file, mapping process now takes longer compare on the time when the message 'No valid key found in chosen key file' is not appearing.

mark-monserrat, 2017-04-25:
As far as I remember, the keys I used for A and B is 000000000000

ikarus23, 2017-04-25:
Hi,
so after you've used the increment function of MCT on sector 1 block 0 you are no longer able to access the whole card? That sounds strange.

Maybe this has something to do with https://github.com/ikarus23/MifareClassicTool/issues/66. However, I was never able to reproduce this behavior.

What device are you using? Do you have access to another device to verify that behavior?

mark-monserrat, 2017-04-26:
Yes, after increment function. But the weird thing is, before it happened, I can use increment/decrement function on the same sector/block without having any problem; and during that time key A & B is FFFFFFFFFFFF, the problem arises when key A and B becomes 000000000000. I'm using xperia XA ultra, android Marshmallow, no firmware upgrade made, the version of the app that I'm using is the latest in playstore as of this writing (2.1.0)

ildar, 2017-04-26:
adb logcat might show something useful

mark-monserrat, 2017-04-26:
Here's adb log on my device:
```
04-26 22:25:33.983 2047-14346/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:33.984 2047-2128/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:33.985 2047-2128/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x3
04-26 22:25:33.989 2047-2128/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:33.992 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:33.996 2047-2128/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:33.998 2047-2126/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:33.998 2047-2126/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x2
04-26 22:25:34.003 2047-2126/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.006 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.010 2047-2126/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.011 2047-2536/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.011 2047-2536/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x2
04-26 22:25:34.018 2047-2536/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.023 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.027 2047-2536/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.028 2047-14346/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.028 2047-14346/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x3
04-26 22:25:34.033 2047-14346/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.038 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.043 2047-14346/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.044 2047-14346/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.044 2047-14346/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x3
04-26 22:25:34.050 2047-14346/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.055 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.060 2047-14346/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.061 2047-2126/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.061 2047-2126/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x2
04-26 22:25:34.068 2047-2126/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.074 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.079 2047-2126/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.080 2047-2536/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.080 2047-2536/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x2
04-26 22:25:34.085 2047-2536/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.090 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.096 2047-2536/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.097 2047-2128/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.097 2047-2128/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x3
04-26 22:25:34.106 2047-2128/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.111 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.111 1166-2240/? I/PowerManagerService: setBrightness mButtonLight 0.
04-26 22:25:34.115 2047-2128/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.116 2047-14346/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.116 2047-14346/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x3
04-26 22:25:34.123 2047-14346/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.126 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.128 1166-1177/? I/PowerManagerService: setBrightness mButtonLight 0.
04-26 22:25:34.131 2047-14346/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.132 2047-2126/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.132 2047-2126/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x2
04-26 22:25:34.137 2047-2126/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.141 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.145 2047-2126/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.146 2047-2536/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.146 2047-2536/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x2
04-26 22:25:34.151 2047-2536/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.154 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.158 2047-2536/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.160 2047-2128/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.160 2047-2128/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x3
04-26 22:25:34.166 2047-2128/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.169 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.173 2047-2128/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.175 2047-14346/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC Key found
04-26 22:25:34.176 2047-14346/? E/NxpExtns: Mifare : phLibNfc_GetKeyNumberMFC returning = 0x0 Key = 0x3
04-26 22:25:34.181 2047-14346/? E/BrcmNfcJni: setReconnectState = 0x0
04-26 22:25:34.183 2047-6437/? E/BrcmNfcJni: getReconnectState = 0x0
04-26 22:25:34.187 2047-14346/? E/BrcmNfcJni: setReconnectState = 0x0
[... 211 lines cut ...]
[... cut ...]
04-26 22:25:34.976 544-544/? I/thermal_src: received msg from kernel, it's len is 28
04-26 22:25:34.976 544-544/? I/thermal_src: TA_DAEMON_CMD_SET_TTJ, ttj 85000
04-26 22:25:34.976 544-544/? I/thermal_src: waiting TA_DAEMON_CMD_NOTIFY_DAEMON from kernel...
04-26 22:25:35.004 2047-7359/? E/BrcmNfcJni: Calling EXTNS_MfcPresenceCheck
04-26 22:25:35.004 2047-7359/? E/NxpExtns: Error Sending msg to Extension Thread
```

The keys in my key file are:
```
FFFFFFFFFFFF 
000000000000 
`````
which are used before the issue occur.

ikarus23, 2017-04-26:
All the things you are describing really sounds like https://github.com/ikarus23/MifareClassicTool/issues/66. You have a Xperia too. I thing this issues is a result of bug with Xperia devices. @alfs said
>However if I reverse the keys:
>000000000000
>FFFFFFFFFFFF
>then MCT now reports the key as being 0s, not Fs.
>If swap the last digit from a 0 to a 1, then the F key is found correctly.
Have you tried "playing" with the keys? (e.g. reverse the order, delete or change a key)

ikarus23, 2017-04-26:
It looks like this is nothing new: https://forum.xda-developers.com/xperia-z2-tablet/help/mifare-classic-support-z2-t2874883

ikarus23, 2017-04-26:
It would be really interesting to see if this issue is fixed in CyanogenMod / LinageOS >= 13.x. @ArchangeGabriel reported in https://github.com/ikarus23/MifareClassicTool/issues/127 that CM fixes the Mifare Classic compatibility issues on the OnePlus One. Maybe it's the same here.

motifs77, 2017-05-24:
Same problem on Nexus 6P😂

motifs77, 2017-05-24:
Clone card, read&write successfully.  Then, can not access the card and same notification like Mark said.

ikarus23, 2017-05-24:
So not exactly the same problem. @mark-monserrat is expecting issues after using the increment/decrement function. Sorry to ask you that: you are sure the keys of the tag you just wrote are in one of the key files you selected, right? Also, there are issues with Mifare Classic and the Nexus 5X. Maybe the 6P has issues as well?! Furthermore, were (some of) the keys of the tag "000000000000"?

osysltd, 2017-05-24:
@mark-monserrat as I can catch from the log
04-26 22:25:35.004 2047-7359/? E/BrcmNfcJni: Calling EXTNS_MfcPresenceCheck

You have broadcom nfc adapter, compatibility for which has not been confirmed probably.

ikarus23, 2017-05-24:
@osysltd good catch! So the Sony Xperia XA Ultra has a Broadcom chip?. Unfortunately I was not able to confirm this by a quick google search.

motifs77, 2017-05-24:
Original card 
"825C7BDB7E0804006263646566676869
00000000000000000000000000000000
00000000000000000000000000000000
000000000000FF078069000000000000

3FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
00455624180511383838383838383838
01170425191432000411A50000000000
000000000000FF078069000000000000

00455622000413180516000000000000
00002359000000000000000000000000
1FFFFFFFFFFEFFFFFFFFFFFFFFFFFFFF
000000000000FF078069000000000000

FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
000000000000FF078069000000000000

00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
000000000000FF078069000000000000"

ikarus23, 2017-05-24:
Interesting. So it seems this is connected to https://github.com/ikarus23/MifareClassicTool/issues/66...

ikarus23, 2017-10-30:
>It would be really interesting to see if this issue is fixed in CyanogenMod / LinageOS >= 13.x. @ArchangeGabriel reported in #127 that CM fixes the Mifare Classic compatibility issues on the OnePlus One. Maybe it's the same here.

Is there anyone with such a device an CM / LinageOS who can test this?

But maybe it is just based on the same consistency issue I have with my Nexus 5X (https://github.com/ikarus23/MifareClassicTool/issues/106). And it looks like Google is responsible for this.

ElPumpo, 2017-12-21:
Yes, CyanogenMod fixed the NFC issues with the OPO a long time ago, this app works perfectly fine on mine. (using sultanxda's lineage 14.1 ROM-9

ikarus23, 2017-12-27:
So it looks like this is an Android issue. I don't think there is anything I can do. Closing this issue until proven otherwise ;).

closed 2017-12-27 by ikarus23