ikarus23/MifareClassicTool #130: cant write  UID ?

xiaodepei, 2017-05-16:
Thank you guys  for sharing this app , and its really nice ,but i met some problems when i use write block0 to change M1 card's UID, iam pretty sure the card i use is the special one which allow me to change uid ,in using this it says success,but actually it didnt the uid is not change any more ...and my device is MI Note 2 and lenove A5500,

osysltd, 2017-05-16:
MCT cannot send special backdoor commands to the card (like pm3 does) for changing UID. MCT can only write to block0 using Mifare Write commands. Not all magic cards accept this way of changing UID. In your case I would suggest using pm with csetuid.

ikarus23, 2017-05-16:
I agree with @osysltd. It is most likely you got your hands on the wrong type of a UID changeable tag.
>The first block of the first sector of an original MIFARE Classic tag is read-only i.e. not writable. But there are special MIFARE Classic tags that support writing to the manufacturer block with a simple write command. This App is able to write to such tags and can therefore create fully correct clones. However, some special tags require a special command sequence to put them into the state where writing to the manufacturer block is possible. These tags will not work.
Remember this when you are shopping for special tags!
Also, make sure the the BCC value is correct before writing. The BCC is the first byte after the UID. It is calculated by XOR-ing all bytes of the UID.

ikarus23, 2017-05-17:
Unfortunately, no. The (official) Android system is not capable of sending such special commands.

osysltd, 2017-05-17:
@ikarus23 cannot find the huge discussion around this, maybe you remember in which issue it was?
In short words we need to rebuild kernel module to be able to send special commands, maybe @xiaodepei will be able to do this.

xiaodepei, 2017-05-18:
@osysltd @ikarus23 Thanks alot !

ikarus23, 2017-05-26:
You are welcome!
I'm closing this now.

closed 2017-05-26 by ikarus23

ArchangeGabriel, 2017-05-26:
@ikarus23 Can you provide links/info about kernel module changes required? Would be nice to get them integrated in LineageOS/Omnirom for instance.

ikarus23, 2017-05-26:
@ArchangeGabriel, sorry, I don't know much about it. I think files like `/system/lib/libnfc...` might be interesting. And of course the kernel module with the driver for the NFC controller (e.g. `pn544`). If you control every aspect of such a chip you should be able to send special commands (e.g. write UID), emulate Mifare Classic tags and crack Mifare Classic tags.

But as far as I know this is a pretty hard job. Mostly because the full documentation of the NFC controllers are confidential. :-/

osysltd, 2017-05-26:
@ArchangeGabriel @ikarus23 https://github.com/ikarus23/MifareClassicTool/issues/67#issuecomment-288194574

ArchangeGabriel, 2017-05-26:
It would indeed be very nice to be able to write UID on such tags but also have those nice features that are emulating tags and mfoc/mfuck like functionnalities. :)

ikarus23, 2017-05-27:
@osysltd oh, yes, I remember this function from the times were I tried to implement custom commands ;) Totally forgot about it. This function should really be a good starting point.

osysltd, 2017-05-27:
I wonder if this is something @LineageOS should look at, at least having some kind of trancieveEx() which then could be fully used in MCT. Also, @ArchangeGabriel might address this to the relevant community for have this feature finally implemented and supported on Android community. This is going to be a breakout implementation.

ArchangeGabriel, 2017-06-02:
Emulation is likely already supported by @LineageOS firmware apparently from some rapid researches (at least it was in CM 9.x). However you can’t have a fixed UID, only random one (and this is apparently hardware).

So there’s probably not a lot of work to do for MCT to be able to emulate tags (with random UID), and likely a bit more for mfoc/mfcuk functions, but should be doable by someone with the knowledge and the time (i.e. not me for both points).