ikarus23/MifareClassicTool #131: issues of checkMifareClassicSupport

zangetsudo, 2017-05-19:
checkMifareClassicSupport  method has two issues(as far as i can tell)

1. the logic in checkMifareClassicSupport doesnot seem right.
please double check the code around nfca.getAtqa.

2. full value of SAK
> SAK is a bit mapping and is recommended to use the bit to check the MIFARE
> type, **the full value of SAK shall not be used to detect a MIFARE type**. For detail check in
> the application note AN10834 - MIFARE ISO/IEC 14443 PICC Selection.

ikarus23, 2017-05-20:
Hi!
1. Sorry but the information you given are not sufficient. What is wrong? Does the algorithm give you false positives? What device are you using? What tag are you using? Is it a tag emulating a Mifare Classic tag? 
2. Good point. I should update the algorithm according to AN10834 page 7. Maybe this will help to improve it.

zangetsudo, 2017-05-22:
code in else imply you are going to do somethings in checkMifareClassicSupport but seems pointless.
they are all return value which is less then zero.

ikarus23, 2017-05-22:
But the big else branch is why I have to do this in the first place. It is not only about knowing if both, tag and device, support Mifare Classic. The other case is just as important. Android is not able to tell you wether it is the device or the that does NOT support Mifare Classic. I need this stuff to display a proper error message ("not a Mifare Classic tag" vs. "Your phone does not support Mifare Classic").

ikarus23, 2017-06-08:
I'm not sure if your observation is general thing of if you actually have a tag that was identified the wrong way. If you actually have such a tag it would be great if you could test the new identify process that I implemented in [this test version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing3.apk).

ikarus23, 2017-10-30:
Any news on this?

ikarus23, 2017-12-27:
Hmm... Still no updates. I hope the changes will work for most people because I will merge them now.

ikarus23, 2018-01-04:
The changes have been released with version 2.2.0 (a44921d5156996d0a6ed1d7e8469fe7a5daefb67). Let's hope that it will work for everybody.

closed 2018-01-04 by ikarus23

ceres-c, 2018-01-07:
It works to me. Tested with a couple of tags and these are the results
- **Mifare Classic**: SAK 08, works as expected with different tags
- **Mifare Plus**: SAK 20, works (there is an [issue ](https://github.com/ikarus23/MifareClassicTool/pull/146#issuecomment-355765257) with a plus in classic-emu mode, but it's not due to this update)
- **Mifare Desfire**: SAK 20, "EV1/EV2" identification is working
- **Standard ISO15693** Not detected, which, I believe, is the correct behavior

I don't have any ultralight at hand, but I'll try and check them as soon as I find where I left those :-)

ikarus23, 2018-01-07:
Thanks for testing!
Unfortunately I don't have any Mifare Plus in SL1 to check what is causing the issue you mentioned. :-/

ceres-c, 2018-01-07:
Don't worry, it looks more like an OS issue...
If you need any further testing contact me :-)

ceres-c, 2018-01-14:
Checked yesterday against two Mifare Ultralight cards: they are detected correctly
I believe you can close this issue now ;-)

ikarus23, 2018-01-14:
That's great! Thanks.