ikarus23/MifareClassicTool #134: Problem with write sector magic 7 byte card.

maxben14, 2017-06-05:
I try write sector in magic 7 byte card, but if i write with option write block this good, but if i want write all sector i get error "Error: Tag lost while checking for keys with write privileges" and if i write only one sector i get error "Error: Some error occurred while writing.This could be really bad."

I think magic card want after each write block do or connect or authentificate in sector with key.
Please, can fix problem.

ikarus23, 2017-06-06:
Have you tried the "Auto Reconnect" option?

ikarus23, 2017-06-06:
Also, there is an authentication before each write block operation.

maxben14, 2017-06-07:
Yes, i try auto reconnect, dont' work. I debug mct and i see that magic card when mct retry authorization in the sector, the card returns no success.

maxben14, 2017-06-07:
This is a problem that the MST is authorized to each block, for example, if it is authorized to 0 block, then it is no longer necessary to re-authorize blocks 1-3, because all these blocks are included in the 0 sector, because of this, the magic card receives the Fail. I try to modify the writeBlock (II [B [BZ) I function, make two calls inside its authenticate (I [BZ)  function and it works.

maxben14, 2017-06-07:
And i have problem with magic card, when i put magic card i see mesage "new tag found uid: 11223344556677" and try click on button OK and it is dont' work only when i check option "Advanced: Enable writing to manufacturer block," Button OK don't work, if i don't check option "manufacter" that button OK work. I try change uid use option Write Block and good change.

Why is not pressed button OK with option manufacturer on magic card and can say me function name ButtonOK_click() how named in source MCT, i want debug this function and find error.
[image]

maxben14, 2017-06-07:
Please do patch in function
private int checkBCC(boolean isWriteBlock) {
......
        int uidLen = Common.getUID().length;
        if (uidLen != 4) {
            // Error. UID is not 4 bytes long.
            return 3;
        }

need replace  do so:
if (uidLen != 4) {
           if (uidLen == 7)
          {
               return 0;
          }
            // Error. UID is not 4 bytes long.
            return 3;
        }

maxben14, 2017-06-07:
My magic card 7 byte, because checkbcc return in my card 3 answer and don't change. I change this function and work now)

ArchangeGabriel, 2017-06-07:
@maxben14 Not related, but where did you get your magic 7 bit card? I’m looking for some too.

maxben14, 2017-06-07:
@ArchangeGabriel on ebay
http://www.ebay.com/itm/7-Byte-UID-Changeable-Card-1K-S50-13-56MHz-ISO14443A/201926495656?_trksid=p2060778.c100290.m3507&_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20160727114228%26meid%3Da8bf5f21d257402787f468fa0a4bc4f1%26pid%3D100290%26rk%3D1%26rkt%3D1%26sd%3D201926495656

maxben14, 2017-06-07:
And one bug when i read magic 7 byte card
[image]
If bits 7C3788 mct don't see keyB, but on other card don't magic read keyB good.
If bits FF0780 mct see keyb in sector. Problem with see keyb in sector only when i change bits on sector.

Can say why mct don't see keyb in sector with bits 7C3788 and how fixed this bug.

maxben14, 2017-06-07:
I do patch in public String[] readSector(int sectorIndex, byte[] key,
            boolean useAsKeyB) 
boolean auth = authenticate(sectorIndex, key, useAsKeyB); replace on

boolean auth = authenticate(sectorIndex, key, useAsKeyB);
if (auth==false)
{
auth = authenticate(sectorIndex, key, useAsKeyB);
}
Now with bits 7C3788 mct read keyb in sectors.

Problem MCT that this tools do authenticate in sector which yet authenticated and magic card on repeat authenticate return false.

maxben14, 2017-06-08:
@ikarus23 can do patch by my information for magic card 7 byte ?

ikarus23, 2017-06-08:
>This is a problem that the MST is authorized to each block, for example, if it is authorized to 0 block, then it is no longer necessary to re-authorize blocks 1-3, because all these blocks are included in the 0 sector [...]

Actually, this is not true. You can set access conditions for each block separately. So block 1 might be writable with key A and block 2 might be writable only with key B. 

> Please do patch in function `private int checkBCC(boolean isWriteBlock)`

Thanks for the bug report. I do not have a 7 byte block 0 writable card and was therefore not able to test it. I will look into it.

>I do patch in `public String[] readSector(int sectorIndex, byte[] key, boolean useAsKeyB)`

That is a strange behaviour...
1. Authenticate the first time --> Successful
2. Authenticate even though the last authentication was successful --> Fail
3. Authenticate again --> Successfull

Right? If so, I think this an issue of the block 0 writable tags. I never experienced this before. Maybe it is best to add an option for such tags.

maxben14, 2017-06-08:
> Right? 
Yes.
I try do option for such card.

maxben14, 2017-06-08:
> a 7 byte block 0
> 
In magic card 7 byte can random bcc byte write and card working. That is, for 7 bytes of cards this check is not needed.

ikarus23, 2017-06-08:
Thanks, but you don't have to. I can do it my own.

ikarus23, 2017-06-08:
OK, so i will fix the check of 7 byte cards.

maxben14, 2017-06-08:
And yet one bug, if i write 2 or more sectors i get error "Error: Tag lost while checking for keys with write privileges". And problem again in authenticate in function:
public HashMap<Integer, HashMap<Integer, Integer>> isWritableOnPositions(
            HashMap<Integer, int[]> pos,
            SparseArray<byte[][]> keyMap)
i do change replace:
if (!authenticate(sector, keys[0], false)) {
                        return null;
                    }
on this:
if (!authenticate(sector, keys[0], false)) {
                        if (!authenticate(sector, keys[0], false)) 
                        {
                             return null;
                       }
                    }

ikarus23, 2017-06-08:
Ok, so when I'm going to implement this option I must consider all authentications MCT does.

maxben14, 2017-06-08:
Yes, correct)

ikarus23, 2017-06-08:
Could you try if the BCC issue is fixed in this [testing version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing2.apk)?

maxben14, 2017-06-08:
Yes, this version change uid, but after successful change uid show error "Error: Some error occurred while writing. This could be really bad." It is error incorect, In fact, the uid has changed successfully

ikarus23, 2017-06-08:
Hmm, this must be due to an error that occoured in `public int writeBlock(int sectorIndex, int blockIndex, byte[] data, byte[] key, boolean useAsKeyB)` (`MCReader.java`). The result of this function was not 0. Maybe again the authentication issue? Can you debug it?

maxben14, 2017-06-08:
Yes, i can debug apk file in ida pro.

maxben14, 2017-06-08:
For debug need in AndroidManifest add android:debuggable="true", i now decompile testing version and add this in manifest and recompile in BatchApkTool.

ikarus23, 2017-06-08:
I already pushed the changes to the master. So you could debug it by using the code and Android Studio.

maxben14, 2017-06-08:
Yes, problem in `public int writeBlock(int sectorIndex, int blockIndex, byte[] data, byte[] key, boolean useAsKeyB) (MCReader.java).`
When MCT change uid it cal writeBlock 2 times, and first call writeBlock return 0 good, and second call return 4. It it problem authentication repeat.
In the android studio did not debug applications, I'm used to using Ida pro for reversing android applications)

maxben14, 2017-06-08:
By the way I did not understand why it is called 2 times, like it is necessary 1 time in fact to cause for change uid 0 block.

ikarus23, 2017-06-08:
Ok, this is strange.  Did you use "write block" or "write dump"?. And it can not be because of the authentication problem because the testing version has no fix for this yet.

maxben14, 2017-06-08:
I use Write dump option.

maxben14, 2017-06-08:
Uid change in magic when mct call first writeBlock function and second call fail but uid yet changed when was first call.

maxben14, 2017-06-08:
If i use option "Write block" then message sucessfull and change uid. In option "write dump" only message about error and uid change too.

ikarus23, 2017-06-08:
Ok, I will look into it.

maxben14, 2017-06-08:
Ok)

maxben14, 2017-06-08:
I do patch MCT apk file in which magic card work good. If you need i can send you.

ikarus23, 2017-06-08:
Could you please try [this testing version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing4.apk)? It does not fix the double write, but it has the retry authentication option.

maxben14, 2017-06-08:
Retry authentication don't work on magic 7 byte in this version. I try 1 sector and 2 and more sectors write but always "Error: Some error occurred while writing. This could be really bad."

maxben14, 2017-06-08:
When i read magic i get https://user-images.githubusercontent.com/26776359/26893318-a4693100-4bc3-11e7-9ce9-c355ae4bf151.png  don't show keyB, it is problem authentification repeat.

maxben14, 2017-06-08:
Please do catlog information about rezult authentification for example "try auth keyB sector X block X fail/good")

ikarus23, 2017-06-08:
Strange. I changed some things in [this version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing5.apk). But there should really be a second try on the authentication.

maxben14, 2017-06-08:
Patch retry work only on sectors which show keyB and don't show ---------. I think i don't patch this https://github.com/ikarus23/MifareClassicTool/issues/134#issuecomment-306898035 because when i read sectors where bits 7C3788 i don;t see keyB and in this sectors don't work write retry patch.

maxben14, 2017-06-08:
Last version retry auth work good)

ikarus23, 2017-06-08:
Nice, Thank you for testing.
* BCC issue --> Done
* Authentication issue --> Done

So "everything" is fine now?

maxben14, 2017-06-08:
Almost, Only when I change uid through an option "Write sector", uid changes and an error show mct.

ikarus23, 2017-06-08:
Oh, yes, of course! Another bug to hunt down.
And I think I will make the retry authentication option disabled by default. Most people do use it and it slows down the key mapping process a lot. But I'm not sure yet...

ikarus23, 2017-06-08:
So what you do is:
* Enable writing to block 0
* Choose "write dump"
* Select only sector 0 (or more?)
* Only UID gets changed and then there is an error

Have you checked if anything other than block 0 was written? E.g. select sector 0 and 1. If I'm not mistaken, only block 0 will get changed and all the other block not. Right?

maxben14, 2017-06-08:
Yes, enable, write dump, only 0 sector (not choose more), only uid change block 0.

maxben14, 2017-06-08:
Yes, right only block 0 change.

maxben14, 2017-06-08:
I think patch retry auth don't work in function `public int writeBlock(int sectorIndex, int blockIndex, byte[] data,
            byte[] key, boolean useAsKeyB)`
need replace if (!authenticate(sectorIndex, key, useAsKeyB)) {
            return 4;
        }

replace so:

```
if (!authenticate(sectorIndex, key, useAsKeyB)) {
   if (!authenticate(sectorIndex, key, useAsKeyB))
{
            return 4;
}
        }
```

ikarus23, 2017-06-08:
Ok, I know the issue now. Somebody found it before: https://github.com/ikarus23/MifareClassicTool/issues/122. Block 0 gets written first. And after a block 0 write command some of the tags disconnect themselves. So I have to write block 0 last.

maxben14, 2017-06-08:
Ok, this error can Ignore, the main uid changes. You do not know in android you can implement a special command on other magic cards that 7 bits use the command?

ikarus23, 2017-06-08:
Sorry, it is not possible to implement the special commands needed to change the other type of UID changeable cards. See https://github.com/ikarus23/MifareClassicTool/issues/82 and https://github.com/ikarus23/MifareClassicTool/issues/102

osysltd, 2017-06-08:
@ikarus23 FYI retry authentication still required for SmartMX in Mifare emulation for example

maxben14, 2017-06-08:
@ikarus23 Do you not know mifare ultralight EV1 Magic cards exist?

osysltd, 2017-06-08:
@maxben14 https://www.aliexpress.com/item/Ultralight-13-56mhz-UID-Magic-keyfobs-tags-writable-free-shipping/32799753183.html

ikarus23, 2017-06-08:
@osysltd Interesting. I thought SmartMX tags work well with only the auto reconnect option.
@maxben14 Have not tried them. But yes, like @osysltd said, there are some UID changeable Ultralight tags available. Not sure if they are EV1 or Ultralight C or something like that.

maxben14, 2017-06-08:
@osysltd it is ultralight basic, about information it is simple ultralight not ev1.

ikarus23, 2017-06-08:
@maxben14 Could you check if [this version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing7.apk) fixes the "some error occoured" error while writing sector 0? I don't think this will fix it, but I least have to try ;)

maxben14, 2017-06-08:
@ikarus23 , "some error" show again when change uid but uid changed)

maxben14, 2017-06-08:
An interesting observation, if I change the 0 block is not the uid, but the end of the 0 block, then the error does not appear.

[24 more replies]

closed 2017-10-30 by ikarus23