ikarus23/MifareClassicTool #135: No valid key found in choosen key file [2]

Johkg011, 2017-06-10:
Hello friend, please help me
I'm cloning a Mifare card, I've used LibNfc and found all the keys. I added them in the MCT Apk and I was able to read the card completely, but at the time of cloning the sectors this error appears .. sectors 1 to 15 were identical, but in sector 0 the KEY B is thus "------ ------ "and with that, Block 0 of this sector is also dashed, I can not clone these sectors 
Please help me
My phone is a Motorola Razr D3
I await an answer

ikarus23, 2017-06-10:
I'm not sure if I understood everything.
1. You "found" all keys with libnfc
2. You were able to read the whole tag with MCT (no dashes anywhere)
3. Sector 1 to 15 were all the same
4. After writing, KeyB of sector 0 is unknown
5. After writing block 0 (of sector 0) is unknown

Right?
What keys did you found? Only "000000000000" and "FFFFFFFFFFFFFF"? What keys are in the file you want to write? Could you give an example?

Johkg011, 2017-06-11:
Exactly.
The keys I found were these:

C620318EF179
8553263F4FF0
0A7932DC7E65
11428B5BCE06
11428B5BCE07
11428B5BCE08
11428B5BCE09
11428B5BCE0A
11428B5BCE0F
18971D893494
44F0B5FBE344
25D60050BF6E
E241E8AFCBAF
8E5D33A6ED51
9F42971E8322
D4FE03CE5B06
D4FE03CE5B07
D4FE03CE5B08
D4FE03CE5B09
D4FE03CE5B0A
25D60050BF6E
E241E8AFCBAF
8E5D33A6ED51
9F42971E8322
7B296F353C6B
3FA7217EC575
D4FE03CE5B0F

The key B of Sector 0 is that 18971D893494, when I read the original card it appears, but when I try to clone that error appears. I already tried to write only Sector 0 and I also could not.

ikarus23, 2017-06-11:
What are the access conditions (part between the keys) of sector 0?

Johkg011, 2017-06-11:
AC's sector 0 = E69871CB

I could see that the Value Block is Sector 0.
Will there be any hidden key that is responsible for the recording?

Johkg011, 2017-06-12:
Friend, I tried to record using a Samsung Galaxy S5 and the problem was the same.
I did the normal reading, but at the time of writing Key B of Sector 0 was "-----", I still checked the option "Advanced: save manufacturer block" but also did not solve.
Please help me

ikarus23, 2017-06-12:
Could you give a full example? (sector 0 should be enough)
* The tag before writing
* The dump you want to write
* The result after writing

Johkg011, 2017-06-14:
I'm cloning a transport card, an original sticker is this:

0195EEDBA18804004323C41000070A0D
ADFDFF7F52020080ADFDFF7F00FF00FF
ADFDFF7F52020080ADFDFF7F00FF00FF
7B296F353C6BE69871CB18971D893494

This is sector 0 of the s50 1k card uid changeable:

A0B115EEEA0804000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF

This is the result after the recording:

--------------------------------
ADFDFF7F52020080ADFDFF7F00FF00FF
ADFDFF7F52020080ADFDFF7F00FF00FF
7B296F353C6BE69871CB------------

Johkg011, 2017-06-14:
The key B and block 0 of this sector are thus "-----"

ikarus23, 2017-06-18:
Ok, so far I have no clue what is causing this. It might be an issue with the block 0 writable tag. Have you tied [this testing version of MCT](http://tests.icaria.de/MifareClassicTool-2.1.0-testing5.apk)?

Johkg011, 2017-07-01:
The problem has been resolved friend.
I had to change the access condition through the hexadecimal calculator.
From E69871 I moved to 19678E, the recording of Block 0 worked.
Thank you.

ikarus23, 2017-07-03:
Oh, ok. Well, if it works now I'm closing this.

closed 2017-07-03 by ikarus23

trustedman, 2020-11-24:
Hello guys.. 

Can help me pls ? https://github.com/trustedman/Mifare-Classic-1k---Money-Read-HELP-ME/issues/1

[image]

[image]

ikarus23, 2020-11-24:
Please don't spam a random issue.
I've replied to your issue.