ikarus23/MifareClassicTool #136: [Feature] Mifare emulation and key recovery

gmkbenjamin, 2017-06-27:
Dear author,
Is it possible to add the below features?
1. Key recovery like mfoc and mfcuk (without using external USB readers)? 
2. Is emulation possible? I know that UID will be randomly generated each time it's read due to Android limitation. But what about other sectors?

ikarus23, 2017-06-27:
Hello @gmkbenjamin, sorry, but it is not possible to implement these features.
1. It is not possible to control the NFC controller with Android in the way it is needed to do key recovery.
2. It is not possible to emulate Mifare Classic tags with Android at all.

ArchangeGabriel, 2017-06-28:
Doesn’t it depend on the driver/chip?

ikarus23, 2017-06-29:
Well, of course you can do both of these things if you have the right chip and the right driver. But the official Android does not ship with drivers that powerful. If we are talking about custom ROMs it should be possible. But I cant't do this things within the code of MCT.

gmkbenjamin, 2017-06-29:
I thought host based card emulation would allow you to emulate ISO/IEC 14443-3 cards. 
https://www.mifare.net/support/forum/topic/creating-nfc-android-app-to-act-as-mifare-card-to-interact-with-mifare-readers/

I will do more digging.

ikarus23, 2017-06-29:
Yes, but Mifare Classic is not compatible with ISO/IEC 14443-3. Only in the anti-collision and selection phase at the very beginning of the communication.

gmkbenjamin, 2017-07-01:
Ah right seems like everything points to driver modification. Thank you for your answer.

closed 2017-07-01 by gmkbenjamin