ikarus23/MifareClassicTool #137: Bluetooth Support

Lostish, 2017-06-29:
Add Bluetooth Support for ACR1255U-J1

As Samsung phones lack support for Mifare Classic ( NFC Cards ) samsung users may want to add support for it via external readers like ACR1255U-J1.

ikarus23, 2017-07-04:
I'm working on supporting external readers like the ACR122U. Well I'm working on supporting the app "External NFC". This app makes it possible to use some USB based NFC reader with Android devices that do not have an NFC reader. Have a look at https://github.com/ikarus23/MifareClassicTool/issues/123 and https://github.com/ikarus23/MifareClassicTool/issues/13.

Well, Bluetooth is another thing than USB. But I don't think I will have the time to implement this. Maybe the developer of "External NFC" is willing to extend his app...

Oh, and MCT is open source. So maybe someone out there has the time and skill to implement this.

sesam, 2018-09-04:
@Lostish see #197 for a possibility. There is even example code, so if you have a special/limited use-case, maybe you or someone you know can take that example code and run with.

ikarus23, 2020-10-16:
Sorry, but I will drop this feature for now.
* I lack the time to implement it
* I want to keep the code clean and open source (no closed source vendor libs)
* Most phones support MIFARE Classic by now (even some old and most new Samsung phones)
* You can get a used (or maybe even new) NFC enabled smartphone that is cheaper than the ACR1255U-J1

closed 2020-10-16 by ikarus23