ikarus23/MifareClassicTool #138: Moto X 2nd Gen - App does not open

dellagustin, 2017-07-05:
I have the following phone:
http://www.gsmarena.com/motorola_moto_x_(2nd_gen)-6649.php
Android 6.0

I installed the app using the PlayStore.
When I try to open the app I get a pop up with the error message "Unfotunately, Mifare Classic Tool has stopped working"

ikarus23, 2017-07-12:
Oh, ok, that is really strange. Have you send the crash report? If so, I can check the stack trace on the Google Play Developer Console.

dellagustin, 2017-07-13:
I just send a crash report, the title was something like crash at start up

ikarus23, 2017-07-13:
Thanks, I will have a look into it. But it might take some time. I will be on vacation ;).

ikarus23, 2017-08-19:
Unfortunately I can not find any crash report of an Moto X 2gen with Android 6.0. But more confusing is, that you was able to install it using Google Play. MCT is blacklisted for the Moto X 2gen because it does not support Mifare Classic (see: https://github.com/ikarus23/MifareClassicTool/issues/1). Are you sure you have a Moto X 2gen?

ikarus23, 2017-10-30:
Any news on this?

dellagustin, 2017-11-01:
Sorry, I lost track of this issue.
I sent a new report.
Here is the about section of my telephone:
[image]

ikarus23, 2017-11-15:
I'm still not sure what might cause this. But you can try [this testing version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing8.apk) and see if anything changes. Please not that this testing version is not finished yet and might behave "strange". ;)

ikarus23, 2017-12-27:
Any news on this? Have you tried out the testing version? Or maybe [this other testing version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing10.apk)?

ikarus23, 2018-05-11:
Lack of feedback. Closed.
Feel free to reopen this after testing the latest version from the Play Store.

closed 2018-05-11 by ikarus23