ikarus23/MifareClassicTool #139: clone card

c64bob, 2017-07-12:
Sorry for what is probably a noob question: I'm trying to clone a Mifare Classic 1k Card, writing only block 0 to a special tag. Most of the dump is being written successfully, but the keys (A & B) from the dump aren't written to the new tag. Shouldn't they be written as well when cloning a tag?

ikarus23, 2017-07-12:
Hi @c64bob,
yes, by using the clone function as much as possible should be written. this includes the keys. However, maybe the tag you want to write on has access conditions that prevent writing the keys.

Can you post the third block (keys & access conditions) of a sector of the target block where the writing fails?

c64bob, 2017-07-12:
Sure:
FFFFFFFFFFFF FF078069 FFFFFFFFFFFF
So it's pretty generic and writing should be possible, right? Can I write just key A for testing purposes on a tag?

c64bob, 2017-07-12:
Actually, just writing block 3 of sector 0 works. Maybe the problem is this: on the tag I'm trying to clone key B is not read and appears as "------------" in the dump. This probably causes the whole block not being written on the new tag. Do you have an idea why key B could not be dumped?

ikarus23, 2017-07-12:
Oh yes indeed, that is correct. Blocks with unknown information will not be written. Mifare Classic only allows writing a full block and MCT can not do that unless the whole block is present in the dump.

c64bob, 2017-07-12:
This makes sense! So why is key B unknown - couldn't it be decrypted, or is it missing from the original tag (if that is even possible)?

ikarus23, 2017-07-12:
MCT can not crack or decrypt keys. It just tries the keys that are in the key file. MCT comes with a bunch of well known keys. However, if your key is not in any of these files MCT can not find it for you.

Have a look at https://github.com/ikarus23/MifareClassicTool#getting-started

c64bob, 2017-07-12:
Thanks for the clarification

closed 2017-07-12 by c64bob