ikarus23/MifareClassicTool #141: More supported models

tiredboffin, 2017-08-12:
UPD INCOMPATIBLE_DEVICES.md: MCT works fine on Google Pixel and on Sony Z5 Compact E5823

jekkos, 2017-08-15:
+1 for google pixel

ikarus23, 2017-08-19:
That is good news. There are still issues on the Nexus 5X... to bad Google does not really care for the soon outdated phone...

I will update the list.

jekkos, 2017-08-20:
Pixel uses pn511 apparently. Too bad none of these chips has a full manual available like the pn532... The only chip that we know how to disable parity check on the protocol

tiredboffin, 2017-08-23:
Isn't it ManualRCVReg (address 1D) Parity Disable Bit 4?

jekkos, 2017-08-23:
So you know how to configure chip to disable parity checks? Might be useful to get mfoc working.. and program Chinese uid cards

ikarus23, 2017-08-28:
You probably still need a custom kernel module to make it work...

jekkos, 2017-08-29:
Hm i thought that the nfc kernel driver was more of a passthrough i2c sink and that most logic is written in userspace? Eg libnfc can do this without installing a custom kernel so not sure if the archirecture for these android NXP chips is that different than from pn532

jekkos, 2017-08-29:
In fact one would need a libnfc port.. this one has some modular drivers for some nxp chips so might be a matter of adding one for android then

ikarus23, 2017-10-30:
Devices added to the list of compatible devices. Closed.

closed 2017-10-30 by ikarus23