ikarus23/MifareClassicTool #143: Format Access Conditions MCT

Johkg011, 2017-08-17:
Hi Ikarus
help me please
I'm trying to format some 1k s50 cards and I'm having problems.
When formatted by the computer through the LIBNFC ​​I can completely erase the card, including the Access Conditions, but by the MCT is not working.
What do I do to format the MCT Access Conditions?

ikarus23, 2017-08-19:
could you paste the content of (at least) one sector of the card you want to format that fails? Also, what device are you using?

Johkg011, 2017-08-19:
The dump of the card I clone has the following Access Condition: E69871CB, this Condition locks the sector, so it can not be formatted to write other data.
But for the notebook I can completely format the Chinese card, I have a file called "format.mfd" that can format the card and I can use it for other recordings.
If I send you this file by email you can not give a solution to solve this problem in MCT?

ikarus23, 2017-08-28:
It looks like you have a special card that allows changing the data or access aonditions although they should be fixed. I didn't even know that these tags exists. Do you use any special program on your computer to format the tag?

Johkg011, 2017-08-28:
So Ikarus, I use the IC 1k s50 UID changeable, by the computer I use an application that binds LIB-NFC, MFOC and MFCUK .. the boy already delivered me with the file "format.mfd" and this file can format 100% of the blocks Regardless of the Access Condition you have in the blocks. But this program can not do anything with the MCT version card, so it would be interesting if you implemented this complete formatting in the app

ikarus23, 2017-10-30:
Sounds indeed interesting. However, to do this, more information is required. How do the programs write these access conditions? Just with the normal commands? If so, I think it should be possible to to the same from a Android phone with MCT. If special commands are needed, I don't think there is much I can do.

Where did you buy that card? If I really want to implement this, I should have at least one test card ;)

Johkg011, 2017-11-23:
Hi Ikarus
 Here is the card link that I use to clone:

https://m.pt.aliexpress.com/s/item/32757074859.html?spm=a2g0n.orderList.0.0.65019c55pS0YOj&

the problem is that this model does not work with the notebook, only by Android.
I used a card from another model and formatted it from the notebook, then read it on the MCT and noticed that everyone has the following data:

D1A4C45FEE08040001B1DEF71535B21D
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF

All cards are unformatted and end with this data in Sector 0 regardless of the Access Condition.

Find a solution for us please

ikarus23, 2017-11-26:
I'm not sure if I get what you are saying. Whats wrong with sector 0 being
```
D1A4C45FEE08040001B1DEF71535B21D
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
```
after format? It looks correct to me. And sector 1 to 15 are
```
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
```
right? This is what a formatted card looks like.

So what is your issue exactly? What is up with two different tags? I thought you have a special tag that can be formatted using libnfc regardless of the Access Conditions. Now you have another tag as well or are we talking about the same tag? And what do you want to achieve? Do you just want to clone a tag?

Johkg011, 2017-11-26:
It's not what I want to know how to totally format a Chinese card by MCT, because depending on the access condition I can not format it fully. Already by the notebook I can format ALL the cards are not important for the condition of access

ikarus23, 2017-11-26:
Can you send me the `format.mdf` via Mail? You can find the address at MCT's Google Play Store page.

rawisha, 2018-01-12:
Hello, Is it possible for you to tell me what software and setup you have for your ACR122U ? the followed softwares that came with is in chinese and weird, i cant really understand any of it .

ikarus23, 2018-01-13:
Also, @Johkg011 how do you exactly use the libnfc tools? Is it something like `nfc-mfclassic W a format.mfd`?!? Or `nfc-mfclassic f a format.mfd keyfile.mfd f`?

ikarus23, 2018-05-11:
Well, there are tags that can be formatted regardless of their access conditions. The Proxmark3 can do the job. However, I haven't had the time to check how it is done. If special commands are needed, it will not work on Android (most likely). If anyone already knows how it is done, I really would appreciate some feedback. I also would appreciate feedback from @Johkg011 regarding  the last questions.

Johkg011, 2018-05-11:
Hi friend
I managed to solve some issues, the card that I recorded for Android and got the dashed line (------) in Key B of Sector 0 was the card known as CUID. But recently they have launched a new model called FUID where all data from Sector 0 appears ..
for now unfortunately I have not been able to format Sector 0 yet when the Access Condition is E69871CB.
Any news I contact

ikarus23, 2018-05-11:
Thanks for the feedback @Johkg011. So it seams these are card specific issues after all.

Johkg011, 2018-05-11:
What is the difference of the normal MCT app for the golden MCT?

ikarus23, 2018-05-12:
From Google Play:
>This is exactly the same version of MIFARE Classic Tool (MCT) as the free one!
(Except it will not show the donate notice at startup/updates.)
>
>If you like MCT and want to support the developer, please consider to buy this app. You will get the exact same version of MCT but in addition, a good feeling for supporting free and open source software ;)

ikarus23, 2018-09-10:
Any news on this?