ikarus23/MifareClassicTool #144: Unable to read content consistently

jimipoh, 2017-09-06:
When I try to read a mifare card with known keys, 9 times out of 10 it will fail (estimation). I also tried advanced Auto reconnect option but it does not make a difference. I am using Huawei P10.

ikarus23, 2017-09-07:
Hmm... So the Huawei P10 has this issue too. I've noticed the same thing for the Nexus 5x. You can find the issue here: https://github.com/ikarus23/MifareClassicTool/issues/106. (Or it is just a tag with a bad antenna.)

As far as I know this is a bug in the NFC stack. I don't think there is much I can do. Well, you can try [this testing version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing5.apk) and see if it helps.

jimipoh, 2017-09-08:
The testing version works. It has been consistent so far. Thanks for the help!

ikarus23, 2017-10-30:
Fix available in options. Closed.

closed 2017-10-30 by ikarus23