ikarus23/MifareClassicTool #148: MCT problem read emulated mifare classic - proxmark3.

maxben14, 2017-10-06:
I try to emulate the mifare classic and read through my smartphone with android MCT apk.
proxmark3> hf mf sim n 0
mf 1k sim uid: N/A, numreads:0, flags:0 (0x00)
#db# 7B UID: 04 53 5d 42 a7 49 80
#db# Reader tried to operate (0x30) on out of range block: 222 (0xde), nacking
#db# Emulator stopped. Tracing: 1 trace length: 27908

proxmark3> hf list 14a
.......
188213078 | 188215446 | Tag | 44 00 | |
188224958 | 188235486 | Rdr | 93 70 88 04 53 5d 82 17 d3 | ok | SELECT_UID
188236978 | 188240498 | Tag | 04 da 17 | |
188243234 | 188245698 | Rdr | 95 20 | | ANTICOLL-2
188247382 | 188253270 | Tag | 42 a7 49 80 2c | |
188257314 | 188267842 | Rdr | 95 70 42 a7 49 80 2c 2d 5e | ok | ANTICOLL-2
188269398 | 188272918 | Tag | 08 b6 dd | |
188503146 | 188507914 | Rdr | 61 00 2d 62 | ok | AUTH-B(0)
188512414 | 188517150 | Tag | 01 02 03 04 | |
188521244 | 188530620 | Rdr |95! cb! 40 8c 0f! 84 ec 75 | !crc| ANTICOLL-2
188539152 | 188543824 | Tag | e7 86 42 2d | |
188614336 | 188619104 | Rdr | 30 de 37 97 | !crc| READBLOCK(222)
188622644 | 188623284 | Tag | 07 | |
188826048 | 188827040 | Rdr | 52 | | WUPA
188828596 | 188830964 | Tag | 44 00 | |
188839586 | 188850114 | Rdr | 93 70 88 04 53 5d 82 17 d3 | ok | SELECT_UID
188851670 | 188855190 | Tag | 04 da 17 | |
188857926 | 188860390 | Rdr | 95 20 | | ANTICOLL-2
188862074 | 188867962 | Tag | 42 a7 49 80 2c | |
188872006 | 188882534 | Rdr | 95 70 42 a7 49 80 2c 2d 5e | ok | ANTICOLL-2
188884090 | 188887610 | Tag | 08 b6 dd | |
188977072 | 188981840 | Rdr | 50 00 57 cd | ok | HALT
................

I try decrypt this with mfkey32
C:\prox\ProxSpace\pm3\tools\mfkey>mfkey64 42a74980 01020304 95cb408c 0f84ec75 e786422d 30de3797
MIFARE Classic key recovery - based on 64 bits of keystream
Recover key from only one complete authentication!

Recovering key for:
uid: 42a74980
nt: 01020304
{nr}: 95cb408c
{ar}: 0f84ec75
{at}: e786422d
{enc0}: 30de3797

LFSR successors of the tag challenge:
nt' : 20f8ed56
nt'': 3c2bcdad
Time spent in lfsr_recovery64(): 0.14 seconds

Keystream used to generate {ar} and {at}:
ks2: 2f7c0123
ks3: dbad8f80

Decrypted communication:
{dec0}: 500057cd

Found Key: [a0a1a2a3a4a5]

I found bug in this command. Why proxmark think that 30 de 37 97 this is decrypt communication ? After authorization, all communication is encrypted. And this is actually the halt command 50 00 57 cd.

And 2 problem with MCT: Error: None of the keys were valid for reading.
The MCT often returns an error when authorizing, when I attach a proxmark simulating a classic card to the phone. I tried it on acr122, everything works well there.

maxben14, 2017-10-07:
Okay, it's good.
But how to deal with the smartphone reader, why he sent a HALT. This is a timings problem, is it possible to somehow speed up the implementation of the algorithm crypto1, for example OpenMP?

ikarus23, 2017-10-11:
Regarding the Proxmark issue: The best place is to discuss this is https://github.com/Proxmark/proxmark3/issues/412

Regarding the read errors:
* It is most likely the Proxmark is emulating something the Android device don't like. There is nothing I can do in MCT. The Android API is responsible of reading the tag.
* Maybe your device has troubles reading Mifare Classic tags. Have you tried the the [latest testing version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing5.apk)? Also, what Android device are you using? And What Android version?

maxben14, 2017-10-11:
My device is sony xperia m2, very good read with mct 1k and jcop card. Android 4.2.2.

ikarus23, 2017-10-11:
Ok, then most likely the proxmark is causing this issues. The simulation commands of the proxmark are not very well know for their stability... e.g. simulating legic tags is broken for years now. Also have a look at https://github.com/Proxmark/proxmark3/issues/105 from 2015...

ikarus23, 2017-10-30:
Most likely an proxmark issue. Closed.

closed 2017-10-30 by ikarus23