ikarus23/MifareClassicTool #149: HELP: Writing a full size (TagMaxSize) byte[]

fi8er1, 2017-10-08:
Hello,

I have been looking for a solution on the internet for quite sometime now and haven't been able to figure out a proper way. 
Need to write some text data (i.e. a user input - 15characters, UserID, IMEI, key) most of it converted to ASCII and a XOR applied. So far, I have been able to write a plain String in the form of NdefMessage and need a way to write the desired text using custom memory mapping (i.e. each desired byte in a specific memory block/page). Moreover, also need to know should I proceed with Ndef or MifareClassic for my purpose.

`for (int i = 14; i < 233; i++) {
                mifare.writePage(i, "    ".getBytes(Charset.forName("US-ASCII")));
                Log.e("pages: " + i, "" + Arrays.toString(mifare.readPages(i)));
                Thread.sleep(100);
            }`
In this code snippet, I'm trying to "clear" pages in loop. However, the loops stops with an error "java.io.IOException: Transceive failed" first on page 5 of the TAG, than on page 13, than again on 96)

I'm new to this and a little help will be greatly appreciated. I have already posted on stackoverflow and haven't been able to get any replies.

ikarus23, 2017-10-11:
I think there are a lot of mix-ups in here:
* Ndef and Mifare Classic are two completely different things. Ndef is a message format. You can put ndef objects (e.g. links, text, contacts, etc.) on any NFC tag. Mifare Classic is a technique. You can put ndef messages on a Mifare Classic tag. Of course you don't have to. You can write any bytes the way you want to your Mifare Classic tag.
* `mifare.witePage` does not sound like Mifare Classic. It sounds more like Mifare Ultralight. These are two completely different techniques and must be used in different ways.
* You "clearing" the pages with `"    ".getBytes(Charset.forName("US-ASCII"))`? It looks like you writing just blanks to every page. Most tags are "clear" if all data is `0xFF` (at least for Mifare Classic).
* Regarding the `java.io.IOException: Transceive failed`: I don't know what is causing thins. Maybe you just don't have the permission to write to this page?!

I hope this helps. You can ask more questions here if you want, but I'm closing this issue because it is off topic.

closed 2017-10-11 by ikarus23

fi8er1, 2017-10-20:
Hey, thanks for your help. I have figured out the techniques, got familiar with the technologies and finally done with the project and I'm happy how it turned out.

However, I accidentally wrote to the restricted pages of my mifare ultralight, which in turn, blocked me from writing to certain other pages as well which were working fine on factory configs. Is there anyway to reset my tag (mifare ultralight) to its default factory state?

ikarus23, 2017-10-20:
Good to hear your project went well!
I'm not very familiar with Mifare Ultralight, but as far as I know, there is not way to reset certain pages  or configurations.